jfedor2/hid-remapper #95: Remapper WebUI unsuable when pairing Bluetooth Game controller

HVR88, 2023-10-21:
Testing with macOS 14.1 (latest RC), webUI running in current version of Chrome.

After successfully pairing a BT game controller, such as XBOX One pad, the mouse cursor begins to move continuously down-left.

This occurs even when there are no mappings and the check-boxes for "Unmapped inputs passthrough on layer" are all disabled.

When previously testing the wired Pico version, un-checking the above boxes seemed to solve the issue.

While the mouse cursor is moving, behavior from clicking buttons on the controller also perform unexpected/odd actions.

HVR88, 2023-10-21:
So, I got the problem to go away by loading/saving the configuration to the board again.

jfedor2, 2023-10-23:
The issue is caused by the fact that mouse cursor movement uses the same usages in USB HID as the left analog stick on a game controller (Bluetooth uses the same standard). Specifically it's caused by the fact that while mouse movement is represented by a signed value (zero meaning no movement), analog sticks are typically represented by an unsigned 0-255 value with 128 being the neutral position (or 0-65535 with 32768 being the neutral position). So if you connect a game controller with passthrough enabled, it results in the mouse cursor quickly going down-right. Wired Xbox controllers are an exception here as they use a signed value with -32768-32767 range and 0 being neutral. An additional issue here is that even if you disconnect the device, HID Remapper remembers the last input state.

Normally a computer distinguishes between these usages by looking at the top level collections they are in (mouse or gamepad), but HID Remapper (currently) ignores those.

HVR88, 2023-10-25:
Thank you for the detailed description. I'll keep an eye on this as I continue to test to see if I ever get into a situation again where disabling Passthrough doesn't work.