jfedor2/hid-remapper #100: Single Pico design not working on USB Power (Works on Dual Pico design)

davidz-yt, 2023-11-16:
I've been testing HID Remapper (r2023-11-08 and 11-15) in a setup where it's connected to just USB Power on the output (instead of connecting it to a PC). Similar to your Xbox Adaptive Controller setup, I'm trying to remap from a USB device to the GPIO outputs. The issue I'm finding is that this does not work on the Single Pico design, but works perfectly fine on the Dual Pico design. Is this expected behavior? 

Both the Single and Dual Pico designs work perfectly with GPIO outputs when the output is also connected to a PC. It's just when you use USB Power (or the PC is asleep) the Single design stops working, but the Dual is fine.

Scenarios Tested:
Working: USB Keyboard -> HID Remapper (Single Pico) -> GPIO Output & PC USB
Working: USB Keyboard -> HID Remapper (Dual Pico) -> GPIO Output & PC USB
**NOT** Working: USB Keyboard -> HID Remapper (**Single Pico**) -> GPIO Output & **USB Power**
Working: USB Keyboard -> HID Remapper (Dual Pico) -> GPIO Output & **USB Power**

jfedor2, 2023-11-16:
I haven't considered this use case before, but knowing how it works underneath it actually makes sense. It's because the remapping engine iterations are synced to different things on these variants, on the dual Pico version they're synced to the USB clock of the host port on the B-side Pico and on the single Pico variant they're synced to the clock of the USB host port of the PC. So when there's no USB connection the remapping engine doesn't run.

I will think about it, it should be possible to make it work.

davidz-yt, 2023-11-17:
Thanks for the explanation. I had a hunch it'd be because of something like that. Regardless, I'm ecstatic it works with the dual Pico variant. Keep up the great work, this solution is amazing!

jfedor2, 2025-03-10:
I know it's been a while, but with the latest release of the firmware you should be able to do this with the single Pico variant.

davidz-yt, 2025-03-10:
Thanks! Just tested it and verified that this works.

closed 2025-03-10 by davidz-yt