jfedor2/hid-remapper #101: Kisnt Mechanical Number Pad, 34 Key will not work with adafruit RP2040 with usb A

Blancits, 2023-11-27:
This seems to work great with a different full size keyboard but when trying with a usb numberpad it seems to freeze when selecting device on the web console. Am wondering if this is an issue with the keypad and if so do you happen to know of something similar that might work as an alternative?

Blancits, 2023-12-06:
Just wanted to follow up asking if there's anything more I can do on my side or is there more information that I could possibly provide you that might be of help.

jfedor2, 2023-12-08:
Can you describe the symptoms again? What do you mean it freezes?

Blancits, 2023-12-08:
First off I'd like to thank you for the work you've done with this, when I had it working for me it was amazing. Originally I had been using your remapper with a spare logitech keyboard I had but the space consumed by having two keyboards on my tiny desk was too much. This is when I decided to buy a 34 key keypad to remedy that. However, the problem that I'm referring to occurs when trying to use that keypad. With the logitech keyboard it was just plug and play but with the keypad after I plug it in and I head over to the remapper config website and select "Open device" (to which then it shows my device) and select my device and click the connect button the browser completely freezes. Due to this I have been completely unable to us the keypad with your remapper. I'd like to thank you once again for taking time out of your day to reply to my post.

jfedor2, 2023-12-11:
Weird!

Not sure what could be causing it. You could try dumping the keyboard's descriptors and some traffic with Wireshark. Start the capture then plug it in (to your PC, without HID Remapper) and press some keys. Might give us some hint.

Blancits, 2023-12-11:
Ok I think this is what you requested but im not too sure. I want to say the keypad is the device that shows as 3.6.2.

[keypad.zip](https://github.com/jfedor2/hid-remapper/files/13630282/keypad.zip)

jfedor2, 2023-12-11:
Yeah, that's it! Let me see if there's anything in there that could confuse HID Remapper...

jfedor2, 2023-12-11:
OK, I think it's this part of the report descriptor that's causing issues:
```
0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
0x09, 0x01,        // Usage (0x01)
0xA1, 0x01,        // Collection (Application)
0x85, 0x06,        //   Report ID (6)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0xFF,        //   Logical Maximum (-1)
0x1A, 0x01, 0x00,  //   Usage Minimum (0x01)
0x2A, 0x0F, 0x04,  //   Usage Maximum (0x040F)
0x75, 0x08,        //   Report Size (8)
0x96, 0x0F, 0x04,  //   Report Count (1039)
0xB1, 0x02,        //   Feature (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection
```
The issue is that we're running out of memory because of the large number of usages. We should probably be handling this better but as a stopgap solution we can give ourselves more memory by disabling copy_to_ram on the single Pico binary variant (historically it was there because of multicore issues when writing config to flash, but I believe that's no longer necessary).

Can you try this test build ("artifact" link at the bottom of the page):

https://github.com/jfedor2/hid-remapper/actions/runs/7170609261

jfedor2, 2023-12-12:
I think the stuck left control thing is in turn caused by this part of the report descriptor:
```
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x06,        // Usage (Keyboard)
0xA1, 0x01,        // Collection (Application)
0x85, 0x09,        //   Report ID (9)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0xE0,        //   Usage Minimum (0xE0)
0x29, 0xE7,        //   Usage Maximum (0xE7)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x75, 0x01,        //   Report Size (1)
0x95, 0x08,        //   Report Count (8)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0xA8,        //   Report Count (-88)
0x75, 0x01,        //   Report Size (1)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0x00,        //   Usage Minimum (0x00)
0x29, 0xA8,        //   Usage Maximum (0xA8)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection
```
Note how it says 0x81, 0x00 - "Input (Data,**Array**, [...]" when it should be 0x81, 0x02 - "Input (Data,**Var**, [...]". Taken at face value this means that every bit of that byte encodes Left Control (if it's zero) or Left Shift (if it's one). Obviously that's not how the keypad works, it's just nonsense in its descriptor. It's mildly interesting that Windows doesn't interpret it this way, which is why it went unnoticed.

Blancits, 2023-12-12:
Oh I see, that explains why the Monitor tab on the configuration shows Left Control every time I press a number key. Luckily this is not much of an issue at all for me. Once again thanks for replying to my issue so quickly and explaining why the issues occurred.