jfedor2/hid-remapper #105: Dosent work with the Truly Ergonomic Cleave

Sentmoraap, 2023-12-17:
The keyboard backlight doesn’t turn on and no keys are sent.

Here is what it sends when I plug it and type a few letters:

[Cleave.pcapng.gz](https://github.com/jfedor2/hid-remapper/files/13697429/Cleave.pcapng.gz)

jfedor2, 2023-12-23:
There's some nonsense in the device's report descriptor, but I don't see anything that would crash HID Remapper. The fact that the backlight doesn't turn on might indicate some hardware incompatibility. Which variant are you using? If it's single Pico then you might have more luck with dual Pico version, but no guarantee of course.

Sentmoraap, 2023-12-26:
Thank you for investigating it.

I am using a single Pico, the Adafruit Feather RP2040 with USB Type A Host.