jfedor2/hid-remapper #106: Unexpected tap hold behavior with macro

rabidalpaca, 2023-12-24:
Is the expected behavior of tap hold where:

Input	Output		Layer	Sticky	Tap	Hold
"1"		"1"			1				x
"1"		"2"			1					x

	Hold returns "2"
	Tap returns "1"

Whereas:

Input	Output		Layer	Sticky	Tap	Hold
"1"		"1"			1				x
"1"		Macro1		1					x
and Macro1 = "2"

	Hold returns "2"
	Tap returns "21"

My setup:
Keyboard: DAS Keyboard 4 Pro
HID Remapper: Adafruit Feather RP2040 USB Host board from Adafruit
UF2:  https://github.com/jfedor2/hid-remapper/releases/latest/download/remapper_feather.uf2
Flashed with: https://www.jfedor.org/hid-remapper-config/

BTW thanks for the great project.  I think adding a modifier support to the output would be a nice feature as well as some of the others tagged feature.  I'm ultimately trying to achieve something similar to tap(1) = "1" and hold(1) = "!".  

Thanks.

jfedor2, 2023-12-24:
Macros aren't currently compatible with tap-hold. I think that could be improved, but for a simple modifier you don't need macros, just make two separate mappings, both with hold enabled:
1 -> 1
1 -> shift

rabidalpaca, 2023-12-24:
Got it, that works!  Thanks!

jfedor2, 2024-01-10:
"Hold" macro mappings should also work correctly now in the latest release.