jfedor2/hid-remapper #108: config-tool-web incompatible version

Famicombro, 2024-01-11:
I'm using the version of two rp2040, and I used Spark (a standalone web server) to run the web configurator for Windows.
The web configurator files were used from the config-tool-web directory. And everything worked great. But after updating the board firmware to the latest (r2024-01-10), it reports the error “incompatible version”. Configurator located at jfedor.org/hid-remapper-config works well.

jfedor2, 2024-01-11:
So update your local instance to the latest version?

Famicombro, 2024-01-11:
I replaced the old files with those located in hid-remapper/tree/master/config-tool-web

jfedor2, 2024-01-11:
... aaaand?

Famicombro, 2024-01-12:
And I received an error message about a version mismatch and wrote here.

Today, later, I will try to flash the firmware again and recheck my web server again. I'll let you know the result.

jfedor2, 2024-01-12:
You already have the latest firmware, you just need the newest version of the configuration website from the repository.

Famicombro, 2024-01-12:
The problem was with my web server. It wasn't picking up the new files. It's working now. I apologize for the inconvenience.

jfedor2, 2024-01-12:
Glad to hear! Sorry for snapping at you!

closed 2024-01-15 by jfedor2