jfedor2/hid-remapper #109: A bit confused about layers

duncanthrax, 2024-01-13:
First of all, thanks for this project. Perfectly executed!

Of course, I have a problem :grin: 

**Trying to map LCTRL-ESC to LALT-F4, while retaining regular LCTRL function (CTRL-C etc.)!**

I start with this:

[image]

Works perfectly, but now all the regular CTRL shortcuts don't work, because LCTRL is never forwarded to the host. I'm aware that I can't add LCTRL passthrough on Layer 1, since that would mean that I end up sending CTRL-ALT-F4. On Linux, that switches to VT4, and I lose sight of my X session. So I'm getting crafty and use ESC to actually switch to Layer 2, and passthrough LCTRL on Layer 1:

[image]

Now I can use CTRL just fine, but when I press CTRL-ESC, I still end up sending CTRL-ALT-F4 instead of just ALT-F4, because both Layer 1 and Layer 2 are active at the same time. 

Is there a solution to this problem? I have experimented with NOTHING, but that does (of course?) not "unset" another keypress.

jfedor2, 2024-01-13:
I don't know if it can be done with layers, but you can do it with expressions.

Make Expression 1 like this:
```0x000700e0 input_state_binary 0x00070029 input_state_binary mul```
and Expression 2 like this:
```0x000700e0 input_state_binary 0x00070029 input_state_binary not mul```

Expression 1 will be nonzero when both Left Control and Escape are pressed and Expression 2 will be nonzero when only Left Control is pressed and not Escape.

Then make mappings like this:

Expression 1 -> Left Alt
Expression 1 -> F4
Expression 2 -> Left Control

duncanthrax, 2024-01-13:
Ah sorry, I had not looked at Expressions yet!

Your proposal works great, just needed a third expression and another mapping to support the case when just ESC is pressed. Thanks!

Could the general mappings config have support for combo inputs? Where the individual keys of the combo would be treated as "unmapped"?

Anyway, I'll close this case.

closed 2024-01-13 by duncanthrax