jfedor2/hid-remapper #111: Easier text writing possible?

fabsenet, 2024-01-19:
Hi,

so first of, I am blown away 🚀  ...this project is so good organised, i found what I needed, i send the gerber files to jlcpcb and a week later, I received 5x v5 devices ready to run and the best part, they actually work! Yay!!

So I want to map a single "useless" key to write a bunch of text like the "hello world" sample. question is simple: is there an easier way than to make a list of single letters? this gets frustrating fast :(

Thx for the project anyway!!!

jfedor2, 2024-01-22:
I'm afraid currently there's no better way. Hopefully in the future.

MGlaab, 2024-05-01:
A work-around option is to export the JSON file. Then modify/write a macro, using a HID Usage Table for values, https://github.com/tmk/tmk_keyboard/wiki/USB:-HID-Usage-Table. Use python JSON to modify the values.

As it turns out I wrote a "Hello World" key remap and it doesn't pickup all the keys that are sent. It gives me "Helo World" when I press my "useless" key. So I'm not sure if longer text strings/macros are supportable yet.

jfedor2, 2024-05-01:
> A work-around option is to export the JSON file. Then modify/write a macro, using a HID Usage Table for values, https://github.com/tmk/tmk_keyboard/wiki/USB:-HID-Usage-Table. Use python JSON to modify the values.

A valid approach!

> As it turns out I wrote a "Hello World" key remap and it doesn't pickup all the keys that are sent. It gives me "Helo World" when I press my "useless" key. So I'm not sure if longer text strings/macros are supportable yet.

When you have two presses of the same key in sequence, you need to put a "Nothing" between them. Otherwise you just get a single slightly longer press.

MGlaab, 2024-05-01:
> When you have two presses of the same key in sequence, you need to put a "Nothing" between them. Otherwise you just get a single slightly longer press.

Now that you say that, I can't believe I didn't realize it. Now I see why you put a "Nothing" in between the "L" in the example.

fabsenet, 2024-05-06:
I really would like to understand, why this is never a problem when doing real typing on real device. A keyboard never lost a keypress from me, as far as  I noticed at least. Do they do something completely different or is it only a matter of me typing toooo slow?

I sometimes write my "HEllo world" (casing intended) inside of a remote desktop and it sometimes behave like missing the pressing of the shift key resulting only in "hello world". any ideas for this?

MGlaab, 2024-05-06:
My guess as to why you'd need to put in extra "Nothing's" is because the code is looking at key presses not at key releases. So in the example "hello world" the computer hears the first L but then the second L isn't heard because technically you never released the L key. So the computer thinks you're holding the L key until It gets the next key press (O). I could be wrong. I'm not a C++ programmer. But key events(key press, key release) are common in other languages.

dglaude, 2024-05-06:
Yes encoding long string of keypress in not really easy, especially if you forget on character. But it is even more difficult when your computer language setting is not QWERTY... you then have to think what QWERTY key to use to have the expected result in let's say AZERTY or QWERTZ. Totally not hid-remapper fault, the keycode are position based and map to EN-US keyboard.

A solution could be a recording option to create macro based on what is typed while in recording mode?

MGlaab, 2024-05-23:
this probably won't be helpful, but I write a python script that takes a string and makes it into a Macro. I'm not a programmer, but it works for me. 
take it for a spin and get a good laugh.
https://github.com/MGlaab/HID-Remapper-Macros-Writer.git

MGlaab, 2024-05-30:
this probably won't be helpful, but I wrote a javascript script that takes an array of 8 strings and makes them into a Macros. It also maps 8 GPIO to those Macros. I'm not a programmer, but it works for me. I've never used Javascript before, but I would like to be able to add a tab in this project that could take textbox input and feed it into my array code and output a JSON file that you could then load or use. If you know JS maybe you could help me add this feature?

take my code for a spin and get a good laugh.

Also, I wrote a python program that does the same thing, 8 strings to 8 inputs.

https://github.com/MGlaab/HID-Remapper-Macros-Writer.git