jfedor2/hid-remapper #112: Switching to a layer exclusively

Kn0rk, 2024-02-14:
So I have the following scenario I have my default layer (0) that includes home row mods. So if I tap the J key it just prints the J key. If i hold the J key it changes to a shift. While this is quite cool sometimes hold behaviour can be troublesome so i have a second layer(2) that can be toggled with the _insert_ key and all it does is remove the hold behaviour.

Finally i have a third layer that i want to use as a navigation layer. This layer should be accessible through the default layer as well as the layer2. With the setup below it works on the default layer. On layer_2 however it will activate layer_3 while layer_2  is still active. So it will print the j and the _return_ character.

Is there a way to prevent this? Can i switch to a layer rather than activating it?
[image]

[sample_config.json](https://github.com/jfedor2/hid-remapper/files/14281292/sample_config.json)

jfedor2, 2024-02-20:
It's not the most straightforward solution, but I think you can implement it like this.

The way it works is it uses Layer 1 as a helper layer and Layer 2 is activated by an expression that says "true if Layer 1 is active, but not Layer 3".
[activate_layer_exclusively.json](https://github.com/jfedor2/hid-remapper/files/14351461/activate_layer_exclusively.json)
[image]
[image]
[image]

Kn0rk, 2024-02-25:
Thanks! I would not have been able to come up with that expression. I just tested it and it works.

closed 2024-02-25 by Kn0rk