jfedor2/hid-remapper #113: Waveshare RP2040-Zero pinouts

davewongillies, 2024-02-15:
I'm attempting to use a Waveshare RP2040-Zero with hid-remapper and wanted to double check the pin outs when using it with hid-remapper as when I connect anything to my RP2040-Zero I don't see anything when I'm on the Monitor tab of https://www.jfedor.org/hid-remapper-config/

Looking at the docs it says:

> D+ to GPIO0 (pin 1), D- to GPIO1 (pin 2), VBUS to VBUS (pin 40) and GND to GND (pin 38)

Looking at the Waveshare RP2040-Zero I think that would that map to?

> D+ to GPIO0 (pin 0), D- to GPIO1 (pin 1), VBUS to VBUS (5V) and GND to GND (GND)

[image]

I've checked continuity from the cable that I've soldered to those pins and that's good. I've tried using the [remapper_waveshare_rp2040_pizero.uf2](https://github.com/jfedor2/hid-remapper/releases/download/r2024-02-14/remapper_waveshare_rp2040_pizero.uf2) firmware file from the last two releases.

jfedor2, 2024-02-15:
Try the regular remapper.uf2 file if you're using pins 0 and 1, the remapper_waveshare_rp2040_pizero.uf2 firmware is for a different board (yeah, their naming is confusing).

bjlaur, 2024-03-14:
I can confirm the remapper.uf2 works with the Waveshare RP2040-Zero when using pins 0 and 1.

MrMennens-tech, 2024-05-22:
> I can confirm the remapper.uf2 works with the Waveshare RP2040-Zero when using pins 0 and 1.

Thanks! Will try this out also!

closed 2026-08-30 by davewongillies