jfedor2/hid-remapper #116: Sticky not working as expected for drag-lock on mouse button

waytgibbs, 2024-03-08:
I am trying to set up hid-remapper so that holding the left trackball button activates drag lock for the right trackball button (via layer 1) and also turns on the scroll lock led on the keyboard while layer 1 is active.
This config:
[image]
does turn scroll lock on and off when the left mouse button is held, but does not make the right button sticky as expected.

This config, with Expression 1 = 1:
[image]
does not turn on the scroll lock LED as expected when the left button is held. (The trackball, not the keyboard, is connected to the RP2040 board; is that why?) This config also does not alter the behavior of the right button as expected.

Am I misunderstanding how layers or the S/T/H flags are supposed to work?

jfedor2, 2024-03-08:
I just tested this configuration and it works as expected with regard to the right button being sticky while left button is held:
[image]
You'd probably also want to add a Left button->Left button mapping with the "tap" flag set so that you can continue to use it as the left button.
Also keep in mind with this configuration, you have to be on layer 1 to "unstick" the right button by pressing it again.
To control the Scroll Lock LED directly, the keyboard needs to be plugged into HID Remapper.

waytgibbs, 2024-03-08:
Thank you for the fast and helpful response! I did not understand that a layer activation works like a momentary switch, not a toggle. A tiny change to the documentation would have prevented my confusion--I'd suggest changing "when" to "while" in the sentence "This means that when that button is pressed, layer X is active and therefore mappings from layer X are applied."

I would add that an option to toggle a layer--i.e. to activate layer 3 with a button tap and keep that layer active until that button is tapped again--would be a powerful new feature for hid-remapper. It would allow users to change how the attached device works for a specific application or class of applications, e.g. drawing apps or certain kinds of games.

jfedor2, 2024-03-08:
You can set the sticky flag on a mapping that activates a layer, making it a toggle.

waytgibbs, 2024-03-08:
Very cool, thanks!