jfedor2/hid-remapper #122: Emulated a single controller using a USB Joystick and a gamepad using free GPIOs

rafik24, 2024-03-21:
Hi Jfedor2, 

I would like to make a hid-remapper controller using an RP2040 with a USB Joystick and a DIY made gamepad.

We can use the current USB interface and add support for buttons using the RP2040 free GPIOs and all this would be seen 
by the Switch as a single controller.

The idea behind this would be to bring a better level of support for players that might need specific hardware made to enable them
to play. Or in other words a much better xbox adaptive controller at an affordable price.

I am a happy to try coding this if I can be pointed in the right direction as I could not put my head around the code and find the right entry points.

xgpt, 2024-03-21:
I've also been interested in this functionality for the same reasons! Feel free to reach out @rafik24 @jfedor2

jfedor2, 2024-03-21:
I don't understand what you're asking. What would you like it to do that's not currently possible?

jfedor2, 2024-03-21:
Specifically, have you seen HID Remapper custom board v6?

https://www.youtube.com/watch?v=3RsDhJDMFjg

rafik24, 2024-03-24:
Hi @jfedor2 it is possible thanks to the amazing work you have done !

I now have my usb joystick and buttons mapping, just struggling with the direction buttons which I hope I will figure out.

Many Thanks !

closed 2024-03-24 by rafik24