jfedor2/hid-remapper #127: Support for the sleep and power buttons

alnat2, 2024-04-16:
@jfedor2 thanks so much for your work on hid-remapper.

As I understand it, the hid-remapper currently has no support for the sleep and power buttons.

I forked the repository and added the report descriptor to the our_descriptor.cc file:

   ```
 0x09, 0x2F,                //   Usage (Phone Mute)
    0x95, 0x01,                //   Report Count (1)
    0x81, 0x02,                //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0xC0,                      // End Collection

    0x05, 0x01,    // Usage Page (Generic Desktop Ctrls)
    0x09, 0x80,    // Usage (Sys Control)
    0xA1, 0x01,    // Collection (Application)
    0x85, 0x04,    //  Report ID (4)
    0x75, 0x02,    //  Report Size (2)
    0x95, 0x01,    //  Report Count (1)
    0x15, 0x01,    //  Logical Minimum (1)
    0x25, 0x03,    //  Logical Maximum (3)
    0x09, 0x82,    //  Usage (Sys Sleep)
    0x09, 0x81,    //  Usage (Sys Power Down)
    0x09, 0x83,    //  Usage (Sys Wake Up)
    0x81, 0x60,    //  Input (Data,Array,Abs,No Wrap,Linear,No Preferred State,Null State)
    0x75, 0x06,    //  Report Size (6)
    0x81, 0x03,    //  Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0xC0,          // End Collection
};
```

After flashing the device with a new uf2 file, hid-remapper no longer forwards inputs to the host.
What did I do wrong?

alnat2, 2024-04-18:
I changed the MAX_INPUT_REPORT_ID to 4. But since I'm not a programmer, I didn't understand in which structure to implement the report descriptor. @jfedor2, would you be able to provide guidance on how to rewrite the report descriptor without using an array and where it should be placed within the file? Thank you for your assistance.

jfedor2, 2024-04-19:
Try adding this to the descriptor:
```
    0x05, 0x01,  // Usage Page (Generic Desktop Ctrls)
    0x09, 0x80,  // Usage (Sys Control)
    0xA1, 0x01,  // Collection (Application)
    0x85, 0x04,  //   Report ID (4)
    0x75, 0x01,  //   Report Size (1)
    0x95, 0x03,  //   Report Count (3)
    0x09, 0x81,  //   Usage (Sys Power Down)
    0x09, 0x82,  //   Usage (Sys Sleep)
    0x09, 0x83,  //   Usage (Sys Wake Up)
    0x81, 0x02,  //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x95, 0x05,  //   Report Count (5)
    0x81, 0x03,  //   Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0xC0,        // End Collection
```

Or see here: https://github.com/jfedor2/hid-remapper/commit/c22ef7f5cc632558a5cbe63fac6f94ea52fd25bf

alnat2, 2024-04-19:
It works, thank you

wayland, 2024-11-02:
Hi @jfedor2 ,

Is there a reason this hasn't been merged?  I ask because I'd like it to be overlooked on purpose instead of by accident :) .  

Thanks!