jfedor2/hid-remapper #129: HID Remapper with hand held speechmike

whyaaronbailey, 2024-04-20:
HID Remapper doesn't seem to recognize or react to button presses from the Nuance PowerMic II. Please see the attached USB capture. I have two projects: one is to remap the buttons from the PowerMic to macros, for which I hope to use HID Remapper. I currently use AHKHID and autohotkey to accomplish this, but I'm looking for a solution that requires no software to be installed on the PC, hence HID Rempapper.  PowerMic API is attached.

I'm also trying to replicate the effects of the "record" and other buttons on the PowerMic using my QMK keyboard, but have not been successful. Any guidance would be appreciated. I've essentially been using a likely erroneous method to do this, which essentially just sends the same HID report as the button press I captured with Wireshark:

`		case RECORD:
        if (record->event.pressed) {
            // Send the HID report for the button press
            uint8_t data_down[32] = {0x1b, 0x00, 0x60, 0x72, 0xe1, 0x28, 0x0a, 0x99, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x09, 0x00, 0x01, 0x01, 0x00, 0x0f, 0x00, 0x81, 0x01, 0x03, 0x00, 0x00, 0x00, 0x00, 0x04, 0x00};
            raw_hid_send(data_down, sizeof(data_down));
        } else {
            // Send the HID report for the button release
            uint8_t data_up[32] = {0x1b, 0x00, 0x10, 0x75, 0x56, 0x2b, 0x0a, 0x99, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x09, 0x00, 0x01, 0x01, 0x00, 0x0f, 0x00, 0x81, 0x01, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00};
            raw_hid_send(data_up, sizeof(data_up));
        }
        return false;  // Skip all further processing of this key`

[2024-0420.zip](https://github.com/jfedor2/hid-remapper/files/15049057/2024-0420.zip)
[Nuance PowerMic API.pdf](https://github.com/jfedor2/hid-remapper/files/15049085/Nuance.PowerMic.API.pdf)

jfedor2, 2024-04-20:
You realize that once you connect the microphone through HID Remapper, it will no longer function as a microphone, right?

whyaaronbailey, 2024-04-20:
Ufff. No I didn’t realize that — I thought pass through would work for everything else :-(

whyaaronbailey, 2024-04-23:
Is it not detected because it comes over as multiple HID devices ( mouse / mic )?

jfedor2, 2024-04-23:
That in itself shouldn't be a problem, HID Remapper should ignore the audio part of the device. It works with a PS5 controller  for example, which also has an audio interface.

I took a quick look at the HID report descriptor in the dump you attached and I don't think there's anything there that would stop it from working either.

Some devices still just don't work. Are you using a single Pico variant of HID Remapper? Dual Pico sometimes works better.