jfedor2/hid-remapper #130: Unable to get board v5 to work

SmoresDecore, 2024-04-20:
Sorry, I dont mean to turn this into a support request but I ordered via JLCPCB 5 v5 boards. 1 was dead on arrival but the other 4 look fine. I am able to flash the firmware on all 4 using remapper_board.uf2 then flash B side but on all 4, no device connected are detected. Once, I managed to get 1 input on the monitor tab but I am unable to reproduce this.

If it helps, I also bought the adafruit board and everything works fine with it.
Any idea what could be the problem?

jfedor2, 2024-04-21:
Hard to say, I may be able to come up with a test firmware that would help determine if B side is even alive (if the problem is with the microcontroller or with the hub chip or something), but to be honest that probably won't help much with making the board work a whole.

jfedor2, 2024-04-21:
I've had a problem with the flash chips in the past that could affect either the A side or the B side and I think the symptoms would match what you're observing, but I made a revision that was supposed to get rid of that issue.

SmoresDecore, 2024-04-22:
I doubt it is something to do with the A or B side. Here is my reasoning:
Ignoring the dead board that does not get detected, all 4 boards were flash with a firmware and I was able to connect to them on the HID page. Soon after, the firmware was updated and when I connected my devices, it warned it had an old firmware. Given all this, I can only assume at least the A side works on all 4 boards. I doubt all B sides are bad though I dont have any way to verify this.

jfedor2, 2024-04-23:
Yeah, if you can flash firmware at all and connect from the web config interface, it means side A is working.

Can you post a closeup picture of one of the boards? Maybe something will jump out.

SmoresDecore, 2024-04-23:
Certainly but it will be in a few days as I am away. Will post update with picture.

SmoresDecore, 2024-04-26:
Alright, here are close up pictures of the boards

Boards 1 to 4
[image]
[image]
[image]
[image]

Bonus Dead Board
[image]

Hopefully this will be helpful.

jfedor2, 2024-04-27:
I though maybe there would be some components missing or soldered in the wrong orientation, but I don't see anything like that.

Try flashing this test firmware, flashing B side, then connecting again and going to the Monitor tab and see if anything shows up there.

[test1.zip](https://github.com/jfedor2/hid-remapper/files/15139531/test1.zip)

SmoresDecore, 2024-04-27:
No luck but I did notice some change.
I am connecting a USB numpad to the HID. With the released firmware, no input appear on Monitor but the light on the numpad is lit to show it is connected. With the test firmware, the light stays off. Well, it briefly flashes as I connect it to the HID but immediately turns off. Tested on 2 boards.

I noticed something. I dont know why but once in a while, I do get 1 input in and it shows on the Monitor but nothing else. It appears the key stays stuck as if held down even after disconnecting the numpad from the HID. Disconnecting the HID "unstucks" the key press.

Just to be sure, I tested the numpad connected directly and it works as expected.

jfedor2, 2024-04-27:
Ah, sorry, that was built for the wrong board, it couldn't have worked.

Try this.

[test2.zip](https://github.com/jfedor2/hid-remapper/files/15139679/test2.zip)

SmoresDecore, 2024-04-27:
Here is what I got. The value keeps going up. So something is working.
[image]

jfedor2, 2024-04-27:
OK, try this.

[test3.zip](https://github.com/jfedor2/hid-remapper/files/15139840/test3.zip)

But after clicking "Flash B side", don't unplug the Remapper, but instead go to the Monitor tab and see what is says.

SmoresDecore, 2024-04-27:
The value got up to 4 then stopped.
[image]

jfedor2, 2024-04-27:
So that at least means that there is communication with the B side. At least the A side seems to think that is has successfully flashed the B side with its firmware.

Maybe there's something wrong with the B-side flash chip.

I may be able to come up with a test firmware where the B-side is be RAM-only, meaning that it isn't stored in flash and only runs once after flashing.

jfedor2, 2024-04-27:
Yeah, just to be clear, best case scenario here is that we identify which part is broken, we will likely not be able to make these boards work with just a different firmware.

SmoresDecore, 2024-04-27:
Yeah, I dont expect these to work anymore but if you want to investigate for whatever reason, I'll help however I can.
I'll see if I can contact JLCPCB and knowing exactly what is wrong might give me a chance to get a refund or some replacement boards.

Linuxmonger, 2025-02-18:
I seem to be in the same boat, I ordered 5 of the v5 boards from JLCPCB and while all of them seem to be working properly, I don't see any indication of communication between the A side and B side RP2040.

Am I just doing this incorrectly? 

The table near the bottom of the first page (jfedor2/hid-remapper/tree/master) indicates that '"Flash B side" required' and I was assuming this meant that after writing 'remapper_board.uf2' to the A side, you then open the online hid-remapper-config site, Actions, Flash B Side and power-cycle. I get a notification that the flash completed and to unplug/replug.

Or, am I supposed to be hooking up a OTG cable to one of the hub ports and flashing there?

This is an amazingly cool project, if there's any chance of recovering these boards, I'm willing to try just about anything, 

Chips all seem to be in the proper orientation and soldered properly;

[image]

Thank you!

jfedor2, 2025-02-18:
> The table near the bottom of the first page (jfedor2/hid-remapper/tree/master) indicates that '"Flash B side" required' and I was assuming this meant that after writing 'remapper_board.uf2' to the A [...]

That is the correct procedure.

What have you tried plugging into it?

Faulty boards happen from time to time, but it would be strange for the faults to only affect the B side.

Linuxmonger, 2025-02-18:
I've tried three different keyboards, Dell, HP. and an old Sun (with 'Again', 'Stop', 'Props', ...keys). They don't seem to power up, no Caps, Scroll, or Num lock.
I just tried a Keychron with backlighting and everything seems to be working, I'm typing this through the hid-remapper! You're a genius!
I swapped the cable between your hid-remapper (HRM) and my keyboard, but I switched everything back and it seems to still work properly.  This is from the Sun plugged directly into the laptop, it doesn't work through the HRM.
If the Keychron is plugged into the HRM and then HRM is plugged into the laptop, it seems to work properly, if I plug a second keyboard in, neither works. Unplugging the second keyboard and then unplugging/replugging the HRM and the Keychron is back on-line.
All of the keyboards work directly, only the Keychron is detected by the HRM, and only if it's plugged into the HRM then that into the laptop.
Maybe needs more power drawn through the ports? I have the backlight pretty bright.
I've got a few dozen KB - and trackballs, I'll try a few others and see what I find out, for now, I'm happy it seems to be working properly at all!
Thank you again!

jfedor2, 2025-02-18:
There might be an issue with Low Speed devices on HID Remappers with the built-in hub (and older keyboards have a good chance of being Low Speed). It's not the hub chip itself I think as it works with those devices when connected to a PC, just in combination with the RP2040. Never got to the bottom of that.