jfedor2/hid-remapper #131: Clear Layer

mreevesh, 2024-04-24:
Fantastic work on this project. 

Would it/is it possible to have a feature that clears all layer selectors, that could be used in a macro. 

I currently have 4 keys set to stick each layer 1-4, but if I’m on a layer, I need to remember which layer I’m on and deselect it before selecting another layer. 

It would be good to have a macro that can deselect all layers, then select the required layer on a single keypress, this would make working with multiple layers easier. 

Thanks

aaronkochar, 2024-06-24:
I agree with this request.  I have been trying to find out if I can use a mouse button to activate Layer 1 and make it sticky, and a keyboard button to release the "stick" of that mouse button (sometimes I go to type and forget to switch Layer 1 off with the mouse button, I'm trying to avoid having to go back to that mouse button to turn off the layer).  I'm not sure if this functionality exists and I'm too non-computer sciency to figure it out; I've looked around and found a couple of things to try and emulate, but had really no understanding of what was happening!

jfedor2, 2024-06-24:
I will keep that in mind in the future.

For now you can implement this functionality using expressions.

Make the following mapping: Register 1 -> Layer 1.

And make an expression like this:
```
1 recall
0x00070004 input_state_binary bitwise_or /* A */
0x00070005 input_state_binary not mul /* B */
1 store
```

This will activate Layer 1 in a sticky manner when the A key is pressed and deactivate it when the B key is pressed.

If you want the A key to also deactivate the layer when pressed again, the expression gets a little more complex:
```
1 recall

/* A pressed and layer not active -> activate it */
0x00070004 input_state_binary
0x00070004 prev_input_state_binary not
mul
1 recall not
mul
bitwise_or

/* A pressed and layer active -> deactivate it */
0x00070004 input_state_binary
0x00070004 prev_input_state_binary not
mul
1 recall
mul
not
mul

/* B pressed -> deactivate layer */
0x00070005 input_state_binary not mul

1 store
```

(I realize it's not the most user friendly thing in the world.)

aaronkochar, 2024-06-26:
Dear jfedor2:

Thank you so much for this solution and your lighting fast response.

I have been using the solution, and though it solved what may seem to have been a minor annoyance, the cumulative effect of being able to use the solution 5k times a day is greatly appreciated!

aaronkochar, 2024-09-30:
I've encountered a new and, I think, related issue.  I don't (yet) use multiple layers at the same time; if Layer 1 is activated, the other layers should not be.  To trigger Layer 1, i use mouse button 7 as a sticky (I press the button once and Layer 1 is active until I press it again or use the above solution to press F1 which also deactivates Layer 1).

My problem is that I hold button 7 to activate Layer 2 and I hold the forward button on my mouse to activate Layer 3.  I'm getting different behaviors if Layer 1 is active or if it is not.  If Layer 1 was not active and I activate either Layer 2 or 3, I get the desired result; if Layer 1 is active, I get the layer 2 or 3 result and the layer 1 result.

Is there a way to turn off all active layers and then activate the one selected layer:
I hit Button 7 and Layer 1 stays on until I hit button 7 again, hit F1, or activate Layer 2 or 3,
I hold Button 7 and if layer 1 or 3 is on, they turn off, and Layer 2 is active (until I let go and it reverts to Layer 0), and
I hold Forward button and if layer 1 or 2 is on, they turn off and Layer 3 is active (until I let go and it reverts to Layer 0)?

Does this make sense, is it possible/easyish?

KillyMXI, 2025-02-01:
> ```
> /* B pressed -> deactivate layer */
> 0x00070005 input_state_binary not mul
> ```

With this, B input is also processed.
I'd like B to do nothing until the layer is cleared.
I tried to add a "B -> Nothing" mapping with layer 1 flag, but that had no effect.
I tried to change order of mappings (a pain to do with no dragability) but that didn't change anything either.
Expressions are processed before mappings, making it impossible?

jfedor2, 2025-02-03:
B is sent because of its mapping on layer 0, not layer 1. Layer 1 is deactivated the moment B is pressed, but it remains pressed for some time after that.

You can make it so that layer 1 is deactivated on B release:
```
0x00070005 input_state_binary not
0x00070005 prev_input_state_binary mul /* B just released */
not mul
```

KillyMXI, 2025-02-03:
This works, thanks.

I suggest enabling the Discussions feature for this repository, so we can have a place for usage questions, configuration recipes, etc.