jfedor2/hid-remapper #132: Support: Startup (run once) action (and Delay)

dglaude, 2024-04-26:
Hello,

This could be a support request, or if there is current way to do this, a feature support.

I try to automatically send a one time key sequence a little bit after computer startup.

(I will use CAPS LOCK, but maybe that work better with SCROLL LOCK)

The first thing is that I would like to verify if the computer is listening to hid-remapper... because sending stuff before that would get lost. So my idea N°1 is to send CAPSLOCK and check if CAPSLOCK LED is turning ON. That is an indication that the computer has finished USB enumeration, understood there is a keyboard and is processing the keys.

So a first layer would be there to send CAPSLOCK keypress... but right now I am floodind so it is not working very well.

Then a second layer would be triggered by using CAPSLOCK LED as input, but I need to figure out if I want to use it as a tap or sticky or hold (watching your video to figure out). The reason for changing layer is that I need to stop sending CAPSLOCK.

Once I am in that second layer, I have to send the key strokes (a macro) that I want to run once.

Then, because I want do that only once, I should switch to a third layer.

Another way I found to do something only once is to do `1 recall` and later `1 1 store` if the result of `1 recall` is zero then that register has not been initialized yet and is still zero (It seems that this is not documented, that the value of an unused register has the value zero).

Any hint, suggestion on how to do this would be great... or if you feel it is not possible yet, please say so, I will save a lot of time not trying.

Thanks

dglaude, 2024-04-27:
One more thing... what I loike about CAPSLOCK and monitoring the LED is that when you connect your hid-remapper, you don't know if it is CAPS or not... so you sequence could do something completely different if you don't check and set.

jfedor2, 2024-04-27:
I'm not sure there's a robust solution to this, but we can of course play around.

To just execute a macro 5 seconds from startup you can make an expression like this
```
time 5000 eq
```
And a mapping: Expression 1 -> Macro 1.

This has one problem, the time counter counter wraps around after like 35 minutes I think?

So we can use a register to hold a "has the macro been executed" flag.

Something like:
Expression 1 (only execute the macro if time=5 seconds and flag=0):
```
1 recall not
time 5000 eq
mul
```
Expression 2 (set flag to 1 if time=5 seconds):
```
1 recall
time 5000 eq
bitwise_or
1 store
```
This should only execute the macro once, 5 seconds from startup.

You CapsLock idea is clever, but you have to ask yourself, do you want the macro to execute after the OS has booted, or in BIOS/bootloader or what. Because those will probably also have a working CapsLock.

But we can try it.

Expression 1 (execute macro if we haven't executed it previously and CapsLock is on):
```
1 recall not
0x00080002 input_state_binary
mul
```

Expression 2 (set register 1 to 1 if CapsLock is on so we know the macro has been executed):
```
1 recall
0x00080002 input_state_binary
bitwise_or
1 store
```

Expression 3 (keep pressing CapsLock every second if register 1 is zero):
```
time 1000 mod 0 eq
1 recall not
mul
```
(Also needs an Expression 3->Caps Lock mapping.)

(Also you'd probably want your macro to start with "Caps Lock", "Nothing", ..., so that Caps Lock is not enabled.)

dglaude, 2024-04-27:
This is really clever... I need to try.

I guess `time 5000 eq` rely on the fact that all expressions are executed every ms... I knew it, but all the example code are blink kind `time 1000 mod 500 gt` so I was not sure I could match a precise value.

Yes I likely have a problem with BIOS then OS... the computer maintain the power, but sure there is a new USB enumeration, so I may have no way to distinguish one event from the other. I need to figure a way.

dglaude, 2024-04-28:
I think we can close this support ticket.

Right now I have a working solution, even if I don't know for sure the detail of inner working.

I did not use the CAPSLOCK trick, nor did I use any layer.

I used twice your trick of doing something only once after a certain among of time:
1. First use is for what need to be done more or less 15 second after power is applied.
2. Second use if for what need to be done more or less 1 minute after power on.

So this is just time based and using a run-once delayed action.
I did not seems to have had problem with the timer wrapping (thanks to the register trick).

I don't know if the step 1 is done once or twice (because of the change from BIOS USB stack to OS USB stack), but it is not a problem since if it append a second time, this is ignored as too early in the boot process.

Thank you very much for the support... maybe you should find another place than github for support, like a discord channel or a forum of some kind? Your YouTube video are really a nice way to introduce concept, but maybe not super interactive.

closed 2024-04-28 by dglaude