jfedor2/hid-remapper #133: How would i? Capslock is Escape, or Control if pressed with a key?

simonvc, 2024-04-29:
I want to do this in hardware. Is it possible?

https://www.autohotkey.com/boards/viewtopic.php?t=92738

jfedor2, 2024-04-29:
Are people really such fast typists? I guess you can always play around with the tap-hold threshold setting.

led, 2024-04-29:
I can only speak for myself, but it's something that I've tried to adjust, but still sometimes find myself coming up against the issue (vim and tmux user, so use caps-lock is used a lot for escape and control - eg this makes the difference from intending to switch tmux panes/windows smoothly (ctrl-a …) but instead ending up just bumping out of insert mode in vim). Really breaks the editing flow.

jfedor2, 2024-04-29:
This wouldn't help with the situation you're describing, but if the problem was that you're getting A instead of Ctrl-A (because you didn't hold CapsLock long enough for the "hold" mapping to trigger) then you could try disabling the "hold" flag and just let CapsLock work as Control all the time. Then if you wanted to press Escape, it would always be preceeded by a short press of Control, but that is usually harmless.

The behavior where the action depends on whether other keys were pressed or not could possibly be emulated with expressions, but that would be some long-ass expressions as they would have to check the state of all the individual keys.

I will keep all this in mind for the future.

simonvc, 2024-04-30:
You guys are awesome. have ordered some hardware.

simonvc, 2024-05-10:
My device arrived. this works perfectly for me:

 CAPS->ESCAPE (tap) 
 CAPS->Left Control (no tap or hold)

closed 2024-05-10 by simonvc