jfedor2/hid-remapper #135: A single key press

MGlaab, 2024-05-08:
Is there a current setting that would allow for a key to be pressed and if it is held down only sends one character as its output?

Example: the left mouse button is mapped to the letter a. The user presses the left mouse button but doesn't release it. The a character continues to be printed to the screen until the user releases it.

The behavior I'm looking for is if the user presses the button and holds it only one a character is printed until released and pressed again.

jfedor2, 2024-05-08:
I think if you make a macro that only sends one letter and map a button to that macro you will get what you describe.

MGlaab, 2024-05-09:
It took me a while to get back to you to test it. Yes this works. 
On my Mac if you press and hold a key it only prints one character. When I tried the old way of mapping a key to a key and then held the key on Windows it repeats that character until release. But your suggestion of mapping a key to a one letter macro works on both systems. Thanks for your help!

MGlaab, 2024-05-09:
I'll consider this closed. Thanks again.

closed 2024-05-09 by MGlaab