jfedor2/hid-remapper #139: having trouble with pairing process, possibly because using a screen reader

a-singer, 2024-06-01:
Hello,

,		I'm sorry to post here, I'm honestly not sure whether these are "issues" or just questions. I'm totally blind and would like to use the remapper. However, I am not sure whether the web-based method to add a bluetooth device, using the Xiao board, is not working, or whether I am just failing to use it properly because I am using a screen reader. When I hit pair, nothing happens. I do not see any sort of interface to select the device I want to pair with and, when I put the BLE keyboard near the board, nothing happens. What is supposed to happen with the Bluetooth remapper? Is there supposed to be an interface which the screen reader is not seeing, or is it supposed to happen automatically in some way? Thanks.

jfedor2, 2024-06-01:
Hi,

I realize this is far from optimal for you, but the way it works is there is no feedback on the web interface when pairing devices.

You click the "Pair new device" button, you do whatever you need to do on the keyboard to put it in pairing mode, usually you hold a button for a moment, and then the rest should happen automatically, the devices should find one another and pair.

Again, I know this isn't helpful when you're blind, but when HID Remapper is in pairing mode, the blue LED is on constantly and when some devices are currently connected, it blinks, with the number of blinks corresponding to the number of devices connected.

I need to think on how to make it all more accessible in the future.

techhitnz, 2024-06-12:
I'm having the same issue.  I can't get a bluetooth keyboard or ps5 dualsense to pair with the xiao.  The blue light stays on solidly the whole time and never pairs with either of those devices.

jfedor2, 2024-06-12:
@techhitnz You're not having the same issue, you're trying to connect a Bluetooth Classic device. Those will never work with an nRF52 based HID Remapper.