jfedor2/hid-remapper #140: inconvenient delay when drag scrolling.

itsnoteasy, 2024-06-04:
i'm using the hid-remapper with the current release and a slimblade pro wired. My settings are to invert the trackball rotation 180 degrees, remap all the buttons to my preferred location, and to slow the drag scroll speed by 0.02. polling rate overclocking is disabled and i use it in the wired mode for 1000hz i think.

my problem is that there is a time lag of 1 to 2 seconds between when i right click to enter layer 1 and when it enters the layer. therefore, i have to wait over the appropriate area, because if i do not, then the mouse moves away when i go to make the drag scroll movement motion, and then i have to repeat the action.

It seems to me to be a bug of some kind since the latency is so low otherwise.

itsnoteasy, 2024-06-04:
[inconvenient_delay.json](https://github.com/user-attachments/files/15571127/inconvenient_delay.json)
here you go.

itsnoteasy, 2024-06-07:
ah that makes sense. forgive me if i have asked about this issue before, which i suspect it wasn't my intention to be repetitive. this should be added to a frequently asked question page.
I think that in theory it should be possible to create a faster layer dedicated to scroll wheel events, because they would not be triggered accidentally due to requiring a specific ball motion in conjunction with the button press. 
This would be similar to the concept of speculative execution in a modern 64 bit cpu.
hid-remapper would enter a layer in parallel with the button tap, perhaps on an extra unused core. 
I will think about it when i am less tired.

itsnoteasy, 2024-06-07:
I managed to fix the issue. I created an extra duplicate layer for drag scrolling, with the only change being that it has the tap flag set, whereas the original drag scroll layer was set to hold. This avoids double scroll events and the lag.
[hid-remapper-nodelay.json](https://github.com/user-attachments/files/15726372/hid-remapper-nodelay.json)

jfedor2, 2024-06-07:
Have your tried a simple "left button -> Layer X" mapping, no tap, no hold?

itsnoteasy, 2024-06-07:
seems to work fine with your suggestion, though i had to add a v scroll to v scroll mapping on layer 1 to make  it less confusing when the wheel didn't work.
I think in older versions, doing this would tend to break the button function. you had to use separate tap and hold to be usable.

closed 2024-06-08 by itsnoteasy