jfedor2/hid-remapper #142: Feature Request: Volume Up/Down Senativity

alvin-000, 2024-06-09:
Request for feature like scaling for mouse that would work for volume control. Instead of moving the default volume up/down interval, it would allow a multiplier (such as x2 or x5 speed). Any help would be much appreciated,I did try to just repeat the keys as a macro, but it did not work.

jfedor2, 2024-06-09:
I just tried it with a macro and it seems to be working correctly.

Just remember to add "Nothing" between the "Volume up" entries.

[image]

alvin-000, 2024-06-09:
Thank you, that was it; I did not put in the Nothing in between and the computer interpreted it as just one long command.

closed 2024-06-09 by alvin-000