jfedor2/hid-remapper #144: Keyboard and Mouse Switch functionality

ams1, 2024-06-14:
Hi,

BTW: I'm loving this repo! 🤗 

I tried the single pico version with an external hub and worked flawlessly.

Have you considered adding the 'Keyboard and Mouse Switch' functionality to hid-remapper?
Ex. ~https://github.com/hrvach/deskhop

I think most of the pieces of the puzzle are all in place - not sure if a digital isolator is mandatory (4 channels, 2 channels...).

I would love to have both remapper and Keyboard and Mouse Switch - I'm guessing you don't have to switch between computers😄.

P.S.: If there's ANYTHING you need help with regarding this repo I have an X handle in the gh profile - feel free to use me 😅.

P.P.S: I'm planning to go down the Pico W Bluetooth + HID Remapper 🐰 hole  - It's my geeky Christmas Wish (hope to get it until... Christmas - I move like a fast 🐌 when it comes to C/C++/hardware)

Thanks.

jfedor2, 2024-06-14:
https://github.com/jfedor2/screen-hopper

ams1, 2024-06-14:
Oh man... How could I have missed this...

Thank you.