jfedor2/hid-remapper #145: Mapping problems with Bluetooth version

ozunion, 2024-06-17:
There may be something I've missed...

The environment used is

> r2024-05-29 remapper_seeed_xiao_nrf52840.uf2
> Bluetooth mouse
> Bluetooth keyboard
> 

The mouse and keyboard are paired with the HID-Remapper and are working perfectly.

When I set "tap-hold: middle button is middle-click when clicked, but switches layer when held" in the example tab, it worked perfectly.
However, when I set "map caps lock to control" while keeping this setting, the caps-lock key remains the caps-lock key.

So I tried adding the "tap-hold~" setting after setting "map caps lock to control" first. 
However, "tap-hold~" did not work properly.
To be more specific, CursorY to Vscroll is not reflected, and the cursor only moves up and down.

I thought that the problem might be with the "Unmapped inputs passthrough on layer" setting, so I unchecked the box, and the cursor stopped moving up and down, but I still couldn't scroll vertically.

So, when used individually, both work well, but when used together, one of them breaks.

Is there something I'm doing wrong with the settings?
Or is this just the way it is?

jfedor2, 2024-06-17:
Can you export your configuration to JSON and post it here and also describe what you expected it to do?

ozunion, 2024-06-17:
This is a JSON file.
If the middle mouse button is held down while moving the mouse, the page will scroll in the direction of the movement (X to H scroll, Y to V scroll).
[hid-remapper-config (1).json](https://github.com/user-attachments/files/15874567/hid-remapper-config.1.json)

I would also like CapslockToControl to work.

jfedor2, 2024-06-17:
The configuration seems correct. Are you sure your keyboard is connected through HID Remapper and not directly to your computer? It may not be obvious with Bluetooth. You can verify that by going to the Monitor tab and pressing some keys on your keyboard. Also the blue LED on HID Remapper blinks as many times as there are devices connected so it should blink twice every second or so.

ozunion, 2024-06-18:
Connections are checked first.The monitor tab shows a report that the keyboard and mouse are each securely connected and recognized to work...

closed 2025-09-14 by ozunion