jfedor2/hid-remapper #146: Re-plug Required When Using KVM

justinlampley, 2024-06-17:
I have been using the single Pico version in conjunction with a KVM since 2022 which allowed me to have the same mappings for the function buttons on my Elecom Huge wired mouse without requiring the identical configuration of the software on multiple computers.  I recently upgraded to the latest version of the firmware, but whenever I use the KVM to switch between computers I now have to re-plug in the mouse in order for any movement or buttons to be recognized.  It appears the issue was introduced with release 2023-07-19, release 2023-07-07 works fine in conjunction with a KVM.

jfedor2, 2024-06-19:
It's probably related to the SOF clock syncing that was introduced between these two versions. Maybe during the switch between the two hosts we stop getting (and therefore sending) SOF packets for long enough for the mouse to suspend or something and we fail to wake it up properly afterwards.

chalermla24, 2024-07-27:
I might be having the same behavior with dual-pico version. Using 3 devices through the usb hub (6gv2 keyboard, trackman marble, wmo). The keyboard will not work (but lights up) when first plugging in the hid remapper while both trackball and mouse works. Requiring re-plugging in the keyboard while the hid remapper is still connected to finally sending input. Interestingly the keyboard will works for a second when the hid remapper is first plug in but will stop working soon after (and will keep sending the same input if a key is hold down when the behavior started).

griffithska, 2024-09-18:
I'm curious if I'm having a similar issue. I just built the 1 pico version last weekend (first time user) with the intention of passing my wireless mouse input through it to my KVM so that I can map a button to Right-Alt so I can switch my KVM using my mouse in addition to my keyboard. This is working about 75% of the time but every once in a while one or both of the connected computers behave as if the Alt key is being held down. Sometimes it will show on the on-screen keyboard, sometimes it won't. So far the fix I've found is that I have to unplug the pico, press some keys on my keyboard and then reconnect.

jfedor2, 2024-12-08:
@justinlampley You can try this test build and see if it makes a difference:

https://github.com/jfedor2/hid-remapper/actions/runs/12225991612

justinlampley, 2024-12-13:
I can confirm that the test build does fix the issue

jfedor2, 2025-03-10:
The latest release contains the fix from the test build above.