jfedor2/hid-remapper #148: Can't monitor joystick X-Axis input from Sega Astro City Mini Arcade Stick

techhitnz, 2024-06-25:
Hi,

I have a Sega Astro City Mini Arcade Stick, and I can monitor every button and joystick input EXCEPT for the joystick X-axis.  Joystick Y-axis can be monitored fine.

[image]

I have tested that the controller is able to be tested fine under Windows (USB controller settings), and with the same USB A to USB C adapter.   

The same USB A to USB C adapter is how I can get my Taito Egret II Mini Arcade Stick working fine on a Nintendo Switch.

Can you please advise.

techhitnz, 2024-06-25:
I have attached a Wireguard USB capture.
[astrocity wireguard capture.zip](https://github.com/user-attachments/files/15979730/astrocity.wireguard.capture.zip)

jfedor2, 2024-06-25:
That device has a broken report descriptor. You can try configuring a custom usage as follows:
[image]

techhitnz, 2024-06-25:
Thank you so much!  That worked.  This is amazing - I can now use controllers that had no 3rd party joystick converter support for on my Nintendo Switch.

techhitnz, 2024-06-25:
Are you able to dumb it down to explain what the issue is and what custom usages are used for.  Forgive me, but I couldn't find any documentation on that specifically.

techhitnz, 2024-06-25:
Also, I wanted to have the configuration to support both my controllers:
 - Taito Egret II Mini Arcade Stick
 - Sega Astro City Mini Arcade Stick

However, now with this new configuration, the Egret no longer functions as expected (compared to when I had it supporting the Egret controller alone).

This is what my mapping looks like:
[image]

Is there any way that I can solve for this through your web configurator.  Or will I need to buy another Waveshare RP2040-PiZero specifically for each controller?

jfedor2, 2024-06-25:
I don't have a great solution to the two controllers needing different configurations.

HID Remapper supports per-device mappings, but it's based on hub ports, not device IDs.

You could probably add a physical switch to your Remapper (just a simple slider or on/off switch wired to one of the GPIO pins) and make it so that the switch activates a layer. Then you could put the mappings for the Egret on, say, layers 0 and 1, and the mappings for the Astro City on layers 2 and 3. Though it might require some modification to the mappings because of the special behavior of layer 0.

For now it's probably simpler to just get another Remapper.

techhitnz, 2024-06-25:
Thank you for taking the time to explain and work through my issue.