jfedor2/hid-remapper #149: Feature Request: Button combinations to activate layers

techhitnz, 2024-06-25:
Hi,

Is it possible to have button combinations activate a Layer?  

For example, I have controllers that are relatively short on buttons.  And rather than assigning a sticky+hold assignment to activate layer, I'd like to be able to activate a layer by pressing a button combination if possible.

Thank you

jfedor2, 2024-06-25:
It's been suggested (#73), but currently it's only possible using expressions.

For example if you make a Register 1 -> Layer 1 mapping and have an expression like this:
```
1 recall

0x000700e0 prev_input_state_binary /* Ctrl */
0x00070006 prev_input_state_binary /* C */
mul
not
0x000700e0 input_state_binary
0x00070006 input_state_binary
mul
mul
1 recall not /* if currently not active */
mul
bitwise_or /* activate layer */

0x000700e0 prev_input_state_binary
0x00070006 prev_input_state_binary
mul
not
0x000700e0 input_state_binary
0x00070006 input_state_binary
mul
mul
1 recall /* if currently active */
mul
not
mul /* deactivate layer */

1 store
```
Then Ctrl-C will activate and deactivate Layer 1.

(It should also be possible with a much simpler expression, because this one completely reimplements the sticky behavior, but I can't get it to work at the moment.)

techhitnz, 2024-06-26:
Thank you.  I managed to get it going with your sample code.  

There's certainly a lot to digest when it comes to working with expressions.  https://github.com/jfedor2/hid-remapper/blob/master/EXPRESSIONS.md#registers.  For the average person, such as myself, do you think there will be an enhancement to the web configurator to do via the GUI

jfedor2, 2024-06-26:
Ideally yes, but no promises as to when.