jfedor2/hid-remapper #154: Devices refuse to work when plugged into USB hub

Kwondor, 2024-07-07:
Hi,

I just recently discovered HID Remapper in hopes that I could use my keyboard and joystick on my Switch.  As such, I bought an Adafruit RP2040 Feather with USB Host and a USB hub.  After flashing it with the latest firmware, I plugged everything in accordingly, first to my PC to set up the mappings.  Nothing showed up under the monitor tab, nor did the RP2040 blink to indicate it was receiving any kind of input.  I did some troubleshooting and plugged in each of my devices separately directly into the RP2040 without the USB hub and they worked/monitored just fine.

I've tried three different USB hubs now - even one that someone else suggested and found success with in another thread - and to no avail.  I'm really at a loss here and looking for some insight.  Thanks in advance!

hanga61, 2024-10-24:
@Kwondor 
Have you found a solution? I´m facing the same issue with the feather board.

Kwondor, 2024-10-30:
@hanga61 No, I gave up.  It looks like I'm gonna have to buy a custom board from JLPCB with the hubs already built in.  I feel like I wasted $20 on the Feather.  It's infuriating.