jfedor2/hid-remapper #155: Feature Request: mouse cursor moves from one monitor to another monitor on correct edge

itsnoteasy, 2024-07-16:
There's an old bug in linux if you have dual monitors and you go fullscreen on youtube, the fullscreen window swaps to the far left monitor, which is awkward if you prefer to watch video on the right monitor. 
The workaround is to set the right monitor as the primary, but then your taskbar moves, but no biggie. the issue comes when you swap the placement of the screens in software xy space so that the mouse moves easily from one monitor to the other, the bug returns and the videos are again on the far left.

so my request is for hid remapper to teleport the mouse to the other edge when it is appropriate. that is to say when the mouse is moved far left it should be on the left screen and when it is on the far right it should be on the right screen, not vice versa. 
i'm running a slimblade pro.

jfedor2, 2024-07-16:
I don't even know how well the absolute mouse mode works on multiple screens, but if it works reasonably then you can probably implement this.

Try the "absolute mouse: basic functionality" configuration from the Examples tab. Just remember to unplug and replug HID Remapper after saving the configuration because it's technically a different emulated device type.

itsnoteasy, 2024-07-16:
the absolute mouse:basic functionality seems to work fine. I wouldn't know where to go from here. I feel bad asking for something that would only suit my needs but if there was a way to remove all borders on a multi monitor setup, so that when you mouse over the edge of a screen you appear at the opposite end of another screen that would be practical and useful.

jfedor2, 2024-07-16:
You can try changing Expression 1 to something like this (second line added):
```
0x00010030 input_state 9 mul 1 recall add 0 32767 clamp dup 1 store
16384 add 32678 mod
```
If I'm not mistaken this should basically swap the screens as far as the mouse is concerned (assuming both screens have the same resolution).

You will probably also want to change the "9" and "16" to some other values depending on your screen resolution. As it is the vertical mouse sensitivity is probably different from horizontal.

itsnoteasy, 2024-07-16:
That works perfectly thanks. It does break the drag scroll setup i had but that's a minor thing. i'll probably use the libinput dragscroll. i happen to have a pair of 16:9 ratio screen so i don't need to change anything i think.
for the expression mapping, is it possible to set the output to a variable, and then another input to the same variable so that i can intercept it for drag scroll.

jfedor2, 2024-07-17:
Can you post the configuration that you were using for scrolling? It's likely they can coexist and it's just a matter of not using the same registers.

itsnoteasy, 2024-07-17:
[hid-remapper-nodelay-v1.json](https://github.com/user-attachments/files/16265894/hid-remapper-nodelay-v1.json)
this is more like spaghetti code to work around a bug where the scrollbar moves under the button. it all works perfectly though, i posted about it in another github issue.
I tried to duplicate the expression as well as changing it's variable 1,2 to 3,4 but it doesn't work. when it changes layer it moves the cursor to the top, and between both screens, when the cursor shouldn't move at all. the only change i made to the expression was to change 9 to -9 and 16 to -16 so as to invert the trackball.
as i say though, it's not important, and ugly hacks like this are usually temporary, so not worth putting effort into.

closed 2024-07-19 by itsnoteasy