jfedor2/hid-remapper #156: Weird Numlock behavior

SmoresDecore, 2024-07-17:
Not really an issue, just something I noticed when using a USB Numpad though HID-Remap.

If Numlock is ON on my main keyboard, the USB numpad's Numlock can be flipped ON/OFF separately, basically giving 2 built-in layers of possible buttons to config with HID. 1 layer with USB Numlock ON and another OFF. Works as expected.

If Numlock is OFF on my main keyboard, the USB numpad's Numlock can be flipped ON/OFF just the same but if a mapped button is pressed while Numlock in ON, it will also turn ON on the main keyboard during the key press.

Just an odd thing I noticed.

jfedor2, 2024-07-18:
What's the setup, the full-size keyboard is connected directly to the computer and the numpad through HID Remapper?

SmoresDecore, 2024-07-18:
Yes exactly

jfedor2, 2024-07-25:
Can you go to the Monitor tab in the web configuration interface and check what the numpad is sending, depending on the NumLock states combination on the full-size keyboard and the numpad?

I'm guessing the numpad must be doing something weird to try and keep an internal NumLock state separate from the OS. A normal keyboard doesn't really care if NumLock is on, it's sending the same keycodes for the numpad keys and just turns the LED on when the computer tells it to.

SmoresDecore, 2024-08-29:
Sorry I didnt reply earlier.
This particular USB numpad seems to have a separate controller for numlock state.
If I have the USB numpad connected to the HID Remapper with NumLock ON and a normal keyboard connected to my PC with NumLock OFF, and press a key, I get this in monitoring. So it is sending a signal to turn on NumLock and the Light.
[image]

I tried to reproduce this effect with 2 normal keyboards and could not replicate this result. The USB numpad could also have separate functions for the same key depending on if Numlock was On or Off. This is not the case for any other keyboards I tried, they all act as if numlock is ON. By that, I mean it registers the "Keypad #" rather than "Insert" or "Home".

My bad, I thought it was a cool quirk of HID but seems like my keypad was just messing with me.