jfedor2/hid-remapper #157: Does HID remapper work with the X-keys L-Trac aux ports?

valdotdev, 2024-07-22:
Hi!

Apologies that this isn't exactly an "issue" but an inquiry.

HID remapper works wonders on my Elecom Huge but I'm also planning to get an X-keys L-Trac but I don't know if the two expandable auxiliary ports at the back can be detected by HID remapper so I'm wondering if that particular trackball feature is supported.

Thanks!

jfedor2, 2024-07-23:
I don't have one, but the buttons you connect to the aux ports on the L-Trac then work as some mouse buttons, right? So if the trackball works with HID Remapper at all then those buttons should work as well. Though they might not be remappable separately from whatever button they work as (if they duplicate an existing button). I'm not sure if those aux ports are configurable on this trackball (when used by itself) or if they have fixed functions.

jfedor2, 2024-07-23:
Oh, and by the way, you can connect external buttons to GPIO pins on HID Remapper itself too! There are even some custom boards with 3.5mm jacks (same as on the L-Trac).

valdotdev, 2024-07-23:
@jfedor2 Thanks for the feedback on the L-Trac! Maybe I'll bite the bullet and try one out.

I'm still new to embedded systems but the GPIO stuff is something I'm very much looking into further in addition to gaining knowledge on that field. Do you have any documentation on how to go about utilizing the GPIO pins for HID Remapper from a beginner's perspective?

Cheers!

valdotdev, 2024-07-25:
Awesome. Thanks for your help!

closed 2024-07-28 by valdotdev