jfedor2/hid-remapper #158: 3D printed case issue with screw holes

TotalCoolness, 2024-07-23:
Im finding an issue with the 3D printed case (single PICO version).  The USB C socket is wedge too tightly between the two screw hole pillars, this causes the screw pillars to become weak and break when you come to putting the screws in. 

Ive broken 3 cases in the exact same spot, the screw pillars just snap off.

jfedor2, 2024-07-24:
A Pico doesn't even have a USB-C socket though?

TotalCoolness, 2024-07-24:
I got it from here. 

https://www.aliexpress.com/item/1005006067143012.html

I take it your 3D print is only good for micro USB ?

jfedor2, 2024-07-24:
I only tried it with the original one from Raspberry Pi, can't really say anything about the clones.

TotalCoolness, 2024-07-24:
OK. 

BTW If any one has a 3D print design for the USB-C clone please drop us the link.

Linuxmonger, 2024-09-13:
Sandpaper, file or knife to scrape away a bit of plastic?

griffithska, 2024-11-05:
> OK.
> 
> BTW If any one has a 3D print design for the USB-C clone please drop us the link.

I was having a similar issue with the case as I bought USB-C variants off AliExpress. I decided to model my own case and it turned out ok. I wasn't quite ready to publish it as its my first model that I've wanted to upload but it is functional. I'll post updates if I get around to refining it.

https://makerworld.com/en/models/759331#profileId-693842
[image]
[image]