jfedor2/hid-remapper #159: Mass production of these

febs, 2024-07-25:
Not an issue, but there used to be a "discussion" section for this project and I can't find it anymore. I did asked this there but it disappeared apparently.

I would just like to know if there is any plan for the author to partner with any aliexpress or similar producer to mass-build these, in a similar way as what happened with GP2040 devices.

I'd love to get a couple of "version 3" PCBs but unfortunately printing those with JLCPCB would still be quite pricey.

Thanks for any replies,

jfedor2, 2024-07-25:
I don't have any current plans, but of course anybody can do it, they don't need my blessing.

febs, 2024-07-26:
Thanks for your feedback.

closed 2024-07-26 by febs