jfedor2/hid-remapper #160: Classic Bluetooth HID Numpad Remapping Keys

gregor202, 2024-07-27:
Hello, 

I want to remap a classic Bluetooth HID Keyboard, but as the Seeed Studio Xiao nRF52840 only supports BLE i would like to know which other hardware i should look for to get this function (in the future)?

BR Gregor

jfedor2, 2024-07-27:
I'm not currently working on making Bluetooth Classic work so I can't really recommend anything.

Possible candidates are the Pico W and ESP32 boards.

tech128, 2024-07-28:
This could be done with a combination of the regular usb hid remapper and a bt adapter that supports hid mode like the DBT120.  Don't use the DBT120 though, it's a royal pain in the butt.  Even when you get it to work once, a lot of the time it does not want to pair again to it for some reason.  Somethin like an orange pi zero 3 would be pretty cheap and reliable to do what you need to do.  Since it can emulate a usb keyboard / mouse when you connect it to a pc from its usb c port.

gregor202, 2024-07-29:
Thank you for your feedback and suggestions! 
I will try using a Raspberry Pi Zero (easier to get locally) to handle the key remapping for my Numpad. 
I will come back how it worked out...

BR Gregor