jfedor2/hid-remapper #161: Feature request: Add Key Cancellation aka. SOCD/Null-cancelling

chalermla24, 2024-07-30:
Is it possible to add this feature? I saw a pull request in qmk git (https://github.com/qmk/qmk_firmware/pull/24000) and thought it might be possible with the hid remapper.

jfedor2, 2024-07-30:
With HID Remapper we don't need to change the firmware for this, it's a matter of simple configuration.

For example for neutral A/D SOCD you could use something like this:

Expression 1 -> A
Expression 2 -> D

Expression 1:
```
0x00070004 input_state_binary /* A pressed */
0x00070007 input_state_binary not /* D not pressed */
mul /* logical AND */
/* used as A */
```

Expression 2:
```
0x00070007 input_state_binary /* D pressed */
0x00070004 input_state_binary not /* A not pressed */
mul /* logical AND */
/* used as D */
```

Here's a JSON file with this config: [socd-neutral.json](https://github.com/user-attachments/files/16423960/socd-neutral.json)

chalermla24, 2024-07-30:
I imported the json file but it doesn't seems to be working.

chalermla24, 2024-07-30:
Could this be because of keyboard having max and min value of 1? I think this is a bug.
[image]

jfedor2, 2024-07-30:
For Last Input Priority you could use something like this (I haven't tested it, but I think it should work):

Expression 1 -> A
Expression 2 -> D

Expression 1:
```
/* register 1 = A has priority */
1 recall
0x00070004 input_state_binary /* A */
0x00070004 prev_input_state_binary not
mul
bitwise_or /* if A was just pressed, A has priority */
0x00070007 input_state_binary /* D */
0x00070007 prev_input_state_binary not
mul
not
mul /* if D was just pressed, A doesn't have priority */
dup
1 store
0x00070004 input_state_binary /* A */
mul /* if A has priority and is pressed then it's pressed */
/* used as A */
```

Expression 2:
```
/* register 2 = D has priority */
2 recall
0x00070007 input_state_binary /* D */
0x00070007 prev_input_state_binary not
mul
bitwise_or /* if D was just pressed, D has priority */
0x00070004 input_state_binary /* A */
0x00070004 prev_input_state_binary not
mul
not
mul /* if A was just pressed, D doesn't have priority */
dup
2 store
0x00070007 input_state_binary /* D */
mul /* if D has priority and is pressed then it's pressed */
/* used as D */
```

[socd-last-input-priority.json](https://github.com/user-attachments/files/16432323/socd-last-input-priority.json)

chalermla24, 2024-07-31:
That works most of the time but doesn't in this specific scenario: Hold A --> Press D --> Let go of D while still holding A
The expected behavior is that the remapper keeps sending A until pressing D, in which it will stop sending A and switch into D instead. That works perfectly. But when you let go of D, instead of going back to sending A because it is still held, the remapper will not send any input anymore.

jfedor2, 2024-07-31:
Ahhh, you're right. I edited the config above, it should now behave as expected.

There's probably a more concise way to do it, I will give it a go later.

chalermla24, 2024-07-31:
Yup that works perfectly. Thank you.

closed 2024-07-31 by chalermla24