jfedor2/hid-remapper #163: Cursor getting "stuck" in corner whenever I use the mouse

jarrodharvey, 2024-08-07:
I've just set up my new v5 board and flashed it with the latest firmware. I haven't changed any settings so far.

When I plug in a mouse (and I have tested with two mice so far) and attempt to move it around for testing purposes the cursor gets "stuck" in the top left hand corner of my monitor. The monitoring tab of the config tool is showing movement on the x and y axis but while I'm moving it around it's like the cursor is "jammed" in the top left hand corner, for want of a better description.

Mice plugged directly into the machine function as normal; this "let me live in the top left hand corner" behaviour only seems to be a thing when I'm using a mouse plugged into my hid-remapper.

jfedor2, 2024-08-07:
Sounds like maybe you changed the emulated device type to "absolute mouse and keyboard"?

jarrodharvey, 2024-08-08:
That was it - thanks jfedor!

closed 2024-08-08 by jarrodharvey