jfedor2/hid-remapper #164: Mapping USB Numpad only when NumLock is OFF

ycl413, 2024-08-11:
I have a USB numpad. I was able to map the buttons successfully. However, it doesn't matter if the Num Lock is on or Off, the same output gets mapped.
Using monitoring, with NumLock ON, when I press Numpad 1, the following are sent:
[image]
With NumLock OFF, when pressing Numpad 1, only the following is sent:
[image]
When I press NumLock alone, nothing shows up on the Monitoring tab.

What I want to do is to only map the keys if NumLock if OFF. If NumLock is on, just let it passthru. Is this possible to achieve?

jfedor2, 2024-08-11:
I think your numpad is trying to maintain a NumLock state separate from the OS. That's why the NumLock key isn't actually sent to the host when it's pressed by a human, but when you press some other key while the numpad's internal NumLock state doesn't match the OS NumLock state (as judged by the NumLock LED state that the numpad is getting from the computer), it first sends a synthetic NumLock keystroke to temporarily toggle NumLock on the computer, then it sends the key you actually pressed, then a synthetic NumLock again to toggle NumLock on the computer. It's weird, but I think I've seen it before.

You can try adding a mapping with NumLock as input and Layer 1 as output, with the sticky flag enabled.

Then just define the mappings you want when NumLock is off normally (on Layer 0).

This should make it so that the mappings are only active when the numpad's internal NumLock state is off and when it's on, everything is passed through as is.

(This will only work if you keep the NumLock state off on the computer at all times.)

ycl413, 2024-08-12:
I added the mapping and it didn't work. It always outputs the mapped keys. Your explanation makes sense, since my main keyboard doesn't have a numpad, the internal state is probably always OFF. I mapped a key on the numpad to output NumLock, and after pressing that key, the behavior on the mapping tab is reversed for the other keys, so it proves that it's trying to compensate the internal NumLock state. 

However, adding your mapping didn't work, it always outputs the mapped key on Layer 0 no matter what the internal state of the NumLock is on. Maybe the numpad is doing something else weird. I think I will just use it as is because I bought it just to map keys, and don't really need the NumPad keys, it's just a nice to have.

Thanks for your help and tool!

dglaude, 2024-08-15:
I have a similar numpad where the numlock key and LED is purely local to the keypad.

I was confused and unhappy.

It send different key code depending on the local state. But some key like "5" might send nothing in non numlock state.
It should be possible to map most key in both state.