jfedor2/hid-remapper #165: Raspberry pi zero support

Wes-T920, 2024-08-13:
More a question than an issue but I couldn't find anywhere more sensible to ask: I have a raspberry pi zero 2W going spare, I was wondering if it might be possible to use that board for this project? 

Obviously the I/O is largely similar to the Waveshare 2040 pi zero but it doesn't seem to be as simple as using that version of the firmware. I'm guessing I'll need to make a few tweaks/add some steps to trick the pi into thinking it's a microcontroller.

If anyone has time I'd appreciate any advice on where to start - or if you think it's a waste of time and I should find something else to do with my spare hardware!

dglaude, 2024-08-13:
I think it's a waste of time and I should find something else to do with my spare hardware!

But that's just my opinion. ;-)

The BCM2710A1 from Raspberry Pi Zero 2W has no relationship at all with the RP2040 used by this project that heavily use PIO to do USB Host.

If you like this project and want to test it, try to acquire a ready made hardware supported.

Wes-T920, 2024-08-14:
I feared that might be the case! I have one avenue to explore and I'll report back if I have any success otherwise I think I might need to buy something to do this project as well!