jfedor2/hid-remapper #166: Virpil Button Panel does not connect

SciFi-Bob, 2024-08-16:
Hi,

I am looking for some debugging options.

The VPC device I have seems to power down if it is not connected to a computer (if I just power it over USB then the lights all come on, few seconds later it goes dark and I get the same behaviour when I connect it to the HID mapper.

I have tested the device with just a vanilla mouse and it works as expected.

Given that there is no software on the PC side that makes this VPC device work, what could be causing the (I assume) no handshake and how can I go about troubleshooting that / getting some screen output as to what is going on. 

TVM

jfedor2, 2024-08-19:
Are you using a single Pico (or something like the Adafruit Feather)? It might work with the dual Pico variant.

As for debugging I normally use a UART-to-USB adapter, though the regular build probably doesn't output anything useful.

jfedor2, 2024-08-20:
Most devices work with all versions, but there are still some that only work with the dual RP2040 variant. Of course there's no guarantee it will work, but there's a chance.

SciFi-Bob, 2024-09-19:
@jfedor2 - Sorry for the long delay!  Life !

Attached is a pcap from the device startup, I hope I did it correctly.
[vpc_cp2_powerup.pcap.txt](https://github.com/user-attachments/files/17067654/vpc_cp2_powerup.pcap.txt)

jfedor2, 2024-09-20:
There's some weird stuff in the report descriptor, but the buttons are there. If it works on PC without any special software then I would expect it to work with HID Remapper. That said, it likely doesn't even get to that point based on what you said earlier, because of some hardware incompatibility (though of course I don't know).

SciFi-Bob, 2024-09-20:
There seems to be a two step process (when observed) a bit like some of the thrust master devices talked about in these comments.  
When you watch it, you plug it in > pause > lights > pause > Dh-Dh from windows.  For hid remepper you get the lights but then they go out again once you would normally get the Dh-Dh!

I don't know what I am really looking at in that pcap really, so I can't peice together what's normal and what's not.

If it helps, I am resonably sure that these sticks are using a micro controler approach to the provide the joystick functions becasuse the included software allows you to program the button behaviour directly into the firmware via a similar flash to your process.

No software is required though PC side.