jfedor2/hid-remapper #167: Activating layers through key combinations

valdotdev, 2024-08-18:
Hi!

Just saw [this](https://github.com/jfedor2/hid-remapper/issues/149) and pretty excited for the eventual implementation.

I believe I also saw somewhere that we can achieve similar functionality via expressions but I just can't wrap my head around the system.

For now I'm trying to activate a layer via LEFT + RIGHT buttons pressed at the same time and map a trackball to make it scroll. So once the layer is activated it's just Cursor Y as input and V scroll as output and optionally make Cursor X do nothing to force it only doing a vertical scroll.

Can anyone guide me on how to achieve this via expressions?

Cursor X = 0x00010030
Cursor Y = 0x00010031
V scroll = 0x00010038
Right btn  = 0x00090002
Left btn = 0x00090001

Thanks!

jfedor2, 2024-08-18:
Let me ask you this first.

When you press the left and right buttons together, you'll never press them at exactly the same time down to a millisecond. What do you want to happen when you press the right button and soon after the left button? Should the right button register as right button or not? When you pressed it, there was no indication whether or not the left button would follow.

valdotdev, 2024-08-18:
Yeah I was actually thinking about that. I wouldn't want to have my right click trigger a context menu event. Left clicking for me is much "safer".

So I was thinking if I do a left click first (it's okay if it registers as a left click initially) then I press the right click button, it checks if the left click is currently pressed down and in turn does not trigger a right mouse click event.

So maybe an order of precedence or prioritizing an initial key first is necessary for this kind of functionality.

For my use case, for example just browsing the web, left clicking to start is okay as long I am aware that I will be doing a key combo and I'll initiate it on a blank part of the screen.

Also just an idea, maybe pressing on a key to activate a layer then pressing another key while holding down the first will trigger a sub-layer of sorts. You will be forced then to initiate a key combo sequentially which for me is perfectly fine.

jfedor2, 2024-08-19:
That's reasonable. And do you want the mouse to scroll only while left+right buttons are held or do you want it to be a toggle (press them once to enable scroll, press again to disable)?

valdotdev, 2024-08-20:
I was thinking the buttons held. But that's because I'm using a trackball so holding the left and right buttons then scroll with the trackball in the middle seems quite natural. But for context I'm using a finger trackball so the ball is on top and between the right and left buttons. A toggle would make more sense with a regular mouse or a thumb trackball.

valdotdev, 2024-08-23:
Just tested it out and it works as expected! Thanks for the trouble and I really appreciate it.

I'm gonna play around with it more and maybe change some stuff if needed (once I get around learning expressions on the device) but right now it's pretty much usable as is for my day to day browsing.

Just got the L-Trac and it is so much of a pain to use try and reach the middle button and scroll wheel on this thing. Other than that the trackball is perfect. 😂

Thanks again! 👍