jfedor2/hid-remapper #170: Command-Line Configureation Tool, No HID Remapper devices found.

nik282000, 2024-08-22:
When using the CLI tool python returns the following:

```
File "common.py", line 127, in get_device
  raise Exception("No HID Remapper devices found.")
```
 
common.py is looking at the results of hid.enumerate() for a device where usage_page == 0xFF00 and usage == 0x0020. When I run hid.enumerate() directly I can see two HID Remapper devices but their usage_page and usage values are both returned as 0:

```
{'path': b'3-7:1.0', 'vendor_id': 51966, 'product_id': 47858, 'serial_number': '', 'release_number': 256, 'manufacturer_string': 'RP2040', 'product_string': 'HID Remapper UC9N', 'usage_page': 0, 'usage': 0, 'interface_number': 0},
{'path': b'3-7:1.1', 'vendor_id': 51966, 'product_id': 47858, 'serial_number': '', 'release_number': 256, 'manufacturer_string': 'RP2040', 'product_string': 'HID Remapper UC9N', 'usage_page': 0, 'usage': 0, 'interface_number': 1}
```

The web configuration tool works as expected after adjusting permissions for the correct /dev/hidraw device. 

I'm running python 3.11.2 on Debian 12 whose current python3-hid package is based on [v0.9.0.post3](https://github.com/trezor/cython-hidapi/releases/tag/0.9.0.post3) of the HID module.

jfedor2, 2024-08-22:
I'm using this "hid" module:

https://pypi.org/project/hid/

Seems like you're using another one?

nik282000, 2024-08-22:
Yes, thank you! The hid module included in the Ubuntu/Debian repos are not compatible.

closed 2024-08-22 by nik282000