jfedor2/hid-remapper #174: Custom usages question

cedartinkerer, 2024-08-31:
When using a custom usage PID/VID/MI etc does this replace what is chosen in the "Emulated device type" drop down or does it add it along side what is chosen in the "Emulated device type" drop down?

In short I have a very picky end device that I believe requires the mouse on MI_00 and I believe TinyUSB puts a mouse at MI_03. 

So I was thinking of using the custom usage to produce an MI_00 device but wasn't sure if it adds it the "Emulated Device type" or replaces it in full. 

Thanks

jfedor2, 2024-08-31:
Custom usages apply to the devices you plug into HID Remapper, they have no effect on what comes out of it.

They are basically a way to act as if the report descriptor of a plugged in device was different than it is (while keeping the actual incoming reports the same of course). They can be used to work around broken or otherwise suboptimal report descriptors.

"MI_00" is just Microsoft-speak for the USB interface of the device. HID Remapper has two, the first is used for mouse, keyboard or whatever else you set the emulated device type to and the second is used for configuration.

So I would expect HID Remapper's mouse to sit on MI_00 and that is indeed what I'm seeing:

[image]

So whatever the issue with your device is, I don't think it's that.

cedartinkerer, 2024-08-31:
Thanks for that information.  That is really helpful.