jfedor2/hid-remapper #176: Command-line Tool

MGlaab, 2024-09-05:
Can I make a request for a YouTube video showing how you can use the command-line tool? I've used the web-based tool, and appreciate it, but I'd like to work up a python automation script that could flash multiple devices at once. (think of a classroom of individual student devices).

jfedor2, 2024-09-05:
You need the Python `hid` module.

There are some OS-specific instructions here: https://pypi.org/project/hid/

I've tried it on Ubuntu and macOS, not sure about Windows.

Then to load a configuration you would do:
```
python3 set_config.py < hid-remapper-config.json
```

This works if there's exactly one HID Remapper connected. If you want to configure multiple devices at the same time you'd have to slightly modify the scripts.

MGlaab, 2024-09-05:
Thanks, I'm going to try with one device on my mac, and see if I can modify it.

jfedor2, 2024-09-05:
On Mac I've had success using MacPorts and installing the module with `sudo port install py-hid`. Just remember to use the Python instance from MacPorts afterwards. If you're using Homebrew it can probably be installed from there as well.

lukeallister, 2024-12-13:
Just commenting on how I got it working on pop-os (ubuntu based).

I needed to create a virtual environment, install system packages, install the
python hid library, and then run the commands with sudo in order to get the right
permissions.
 
install packages 

```
sudo apt install libhidapi-dev libhidapi-hidraw0
```

Use micromamba to create an environment (could do this with conda or venv)

```
micromamba create --name hid python=3.12
micromamba activate hid
pip3 install -r requirements.txt
```

Run the `get_config.py` script as root and output the config. Edit the config,
then write back to the device.

```
sudo /home/luke/Dev/micromamba-env/envs/hid/bin/python3 get_config.py > config.json
vim config.json
cat config.json | sudo /home/luke/Dev/micromamba-env/envs/hid/bin/python3 set_config.py
```

Is there a good documentation of what the 'target_usage' and 'source_usage' mean? I just had to remember how I'd set it on the web config from a windows device.

jfedor2, 2024-12-13:
Some labels are in usages.js, in general they're defined here:

https://usb.org/sites/default/files/hut1_5.pdf