jfedor2/hid-remapper #177: Time limits  on button activation

dehrhart, 2024-09-05:
Currently I use switch modifiers from EnablingDevices to limit the activation period of a button presses.  It really helps helps students with motor processing disorders not spin in circles in games like MarioKart and Minecraft. It would be great to be able to do this with HID Remapper on a Adafruit feather (smaller and less expensive)  for buttons and even the x y axis.
Thanks!

[image]

jfedor2, 2024-09-09:
Here's a configuration that applies a 1 second timeout on all buttons and the d-pad:

[hid-remapper-gamepad-timeout.json](https://github.com/user-attachments/files/16935575/hid-remapper-gamepad-timeout.json)

I hope I understood your question correctly. This is assuming you're plugging the Flex into HID Remapper and HID Remapper into the Switch.

The implementation is different for the buttons and for the dpad. For the buttons it uses HID Remapper's built-in "hold" state and the timeout threshold is controlled by the "tap/hold threshold" setting in the Settings tab. For the d-pad it's implemented manually with an expression (change the "1000" (milliseconds) value in Expression 2 to change the timeout).

I admit that this isn't very straightforward, but there is no built-in way to do it.

The dummy mappings with "Nothing" as output are there just so that HID Remapper knows to keep the "hold" state that is then used in Expression 1. They do nothing by themselves.

If you don't want the timeout behavior on some button, just delete the two mappings that mention it as input or output.

If you want different timeout values per button then each would have to be implemented similarly to the d-pad.

dehrhart, 2024-09-13:
Thanks! I'll definitely be trying it out!