jfedor2/hid-remapper #178: NiceNano clones connectivity issue

JostBrand, 2024-09-12:
Hi,

First, I want to thank you for this awesome project. I am running a couple of split keyboards with NiceNano clones and wanted a solution to enter LUKS passwords wirelessly. The process for setting up your tool on an additional nRF52840 and pairing my keyboard took less than 3 minutes. 

**Problem:**

I have noticed a clear connectivity issue when typing. Some keystrokes are just not recorded. When connecting to my regular Bluetooth dongle, I have no issues.

**Questions:**

1. Are these chips missing anything required to have a stable connection with this tool? 
2. Or is it more likely that the board definition of the `Adafruit Feather nRF52840 `is a little bit different?

Thanks for your help in advance.

jfedor2, 2024-09-12:
Sorry, what are you running HID Remapper on? Another NiceNano clone? If it's running at all then I don't think the board definition would be causing dropped inputs.

And what are you running on your keyboards? ZMK?

JostBrand, 2024-09-13:
Sorry for not providing more context. The [nice!nano](https://nicekeyboards.com/nice-nano/) is a well known Nordic nRF52840 chip used in DIY Ergonomic keyboards.  Here is a popular example [Sweep](https://github.com/davidphilipbarr/Sweep). Most of them have two of these chips (one for each side) one connecting to the other half, while the main half connects to the PC.

There are different firmware options to run these keyboards like QMK, KMK and ZMK. Where [ZMK ](https://zmk.dev/)is the most used option for wirless keyboards. They allow to fully customize the keyboard. 

So I have one NiceNano on my keyboard running zmk and one with your software running connected via usb to my desktop pc as a dongle.

domiluci, 2024-09-23:
@JostBrand Which nice!nano clone are you using? I've tried with the [Pro Micro](https://keeb.io/products/supermini-nrf52840-pro-micro-bluetooth-le-ble-controller) "nice!nano" version, using the Adafruit Feather nRF52840 version of HID Remapper. And while it shows up as an HID Remapper device and accepts programming, and even connects to my Bluetooth devices and accepts their inputs, nothing gets output by the hardware via USB. Joypad.ai shows nothing when I test the mapping. 

It sucks because I have a ton of these I got dirt cheap, and my XIAO nRF52840 Sense isn't something I want to use for this long term, nor buy another of considering one costs 4-8x as much as one Pro Micro nRF52840 that I have in oodles.

Oh, and the LEDs are kind of wonky on the Pro Micro as well. Red LED doesn't work, and Blue LED just flashes no matter what.

So I'm just curious how you got these working. Thanks!

jfedor2, 2024-09-24:
@domiluci Can you post your configuration (JSON file)?

domiluci, 2024-09-24:
@jfedor2 Sure thing, though it's not something I'm particularly proud of. The expressions are a bit confusing, so I basically just went with trial and error until it functioned how I wanted it to. And it does function perfectly fine when I run it on my XIAO nRF52840 Sense. I can use my Bluetooth trackpad as a Left Stick that resets back to center when I lift my finger up and off it, which is what I wanted. But when I flash the nRF52840 Pro Micro with the firmware, then connect the same BT trackpad and load this same config onto it, everything is mission go, but there's no output being registered from the Pro Micro despite the fact I can see the exact same input from the BT trackpad under the "Monitor" tab as I do with the XIAO.

Everything is essentially identical between the XIAO and Pro Micro, sans the LEDs, but there's no output from the Pro Micro.

I've got to run, but I'll pick this back up when I get home. Thanks!

JSON Config: [20240922-P1MouseBT_to_LeftStick.json](https://github.com/user-attachments/files/17115622/20240922-P1MouseBT_to_LeftStick.json)

jfedor2, 2024-09-24:
While I don't see how this would explain the behavior you're seeing, you generally don't want two expressions setting the same output.

Instead you can have one expression that sets it to the value coming from the trackpad coordinates when your finger is on it and 128 otherwise, like so:
```
0x00010030 input_state 0 add 0.099 mul
0x000d0042 input_state_binary
mul

128
0x000d0042 input_state_binary not
mul

add
```

And similarly for the Y axis. Here's the whole config: [trackpad-to-stick.json](https://github.com/user-attachments/files/17117375/trackpad-to-stick.json)

Like I said I don't think this would explain your issue, but hey, you can give it a try.

(I assume you're seeing the inputs from the trackpad in the Monitor tab on the "Pro Micro" board?)

jfedor2, 2024-09-24:
@JostBrand Sorry, I missed your edit with further info. That is interesting. Something like that could happen if both the key-down and key-up reports somehow got batched from HID Remapper's point of view and the processing engine didn't run in between them. (Then the next time it runs, the key is not pressed and therefore the keystroke is missed.)

I'll try to come up with something to verify this hypothesis.

jfedor2, 2024-09-24:
@JostBrand One thing you could check is do these missed keystrokes show up in the Monitor tab on the configuration website?

domiluci, 2024-09-24:
@jfedor2 I was 100% sure I had the expressions gorked up in some way LOL. Like I said, I found it a tad confusing, so I just trial-and-errored it until it did what I wanted. Thank you very much for fixing it though. That was very kind of you 😄.

And it actually seems to have fixed the problem because now Joypad.ai is registering output from the nice!nano/Pro Micro. Why my bad expressions worked on the XIAO and not the Pro Micro when they're functionally identical was beyond me. But then I realized (and I'm spitballing as I haven't looked through the code or repo extensively) that they may actually be somewhat different, code-wise, as the Pro Micro takes the Adafruit Feather nRF52840 firmware while the XIAO nRF52840 takes the XIAO nRF52840 firmware. Like I said, I'm not well versed on your code or repo, so I'm hoping I'm not making an idiot of myself here, but maybe there's a difference in the two firmwares that made the Adafruit Feather nRF52840 firmware on the Pro Micro not pass my messed up expressions through? Just a thought. Now if I could just get that RP2040 Pi Zero working... 😆

Anyways, sorry for hijacking the thread. But thank you for the help on this. It's very much appreciated!

jfedor2, 2024-09-25:
@JostBrand I made a test build of the firmware that addresses what I think might be the issue causing dropped inputs. You can try it here (the "artifact" link at the bottom):

https://github.com/jfedor2/hid-remapper/actions/runs/11039276703

JostBrand, 2024-09-26:
@domiluci I used the superMini ones, I think these are the most popular on Aliexpress.

@jfedor2 Thank you, your test build fixes the issue now the &mt are registered as they should. :heart_eyes:

closed 2024-12-14 by JostBrand