jfedor2/hid-remapper #179: Using 2.4Ghz wireless devices?

CrowKing63, 2024-09-21:
As the title says, I want to know if it is possible to have a wireless connection via a 2.4Ghz adapter. 
I've actually tried it, I've tested both a bluetooth dongle adapter and a 2.4Ghz wireless adapter, but I wasn't expecting anything from the bluetooth, and I was expecting something from the 2.4Ghz wireless, and they both failed.
The reason I'm posting this question is because I've confirmed that the 2.4Ghz wireless communication works with a device called the Titan Two, so I have a small hope that it's just not working due to device compatibility, and that the 2.4Ghz adapter itself works. 
So I thought I'd ask just to be sure.

CrowKing63, 2024-09-21:
Aha! Good to know that it works. Thanks for letting me know.

The product I tried, VXE R1 SE PLUS, is a recently released cheap mouse from China. I plugged in the adapter, moved the mouse around and even clicked on it, but nothing came up in the Monitor tab.

CrowKing63, 2024-09-28:
hello. I once sought your advice to remap the joystick in 8BitDo Lite SE to an 8-way switch, and you wrote the code to make it accessible, which I am very grateful for. 

Recently, 8BitDo released the second generation of the Lite SE with Microsoft. I bought it because I liked the ability to add external switches, and I tried to remap it once again, but the new gamepad is not recognized. Both the 2.4Ghz wireless connection and the USB cable wired connection are not recognized. I don't see anything in the monitor tab. Can you give me another idea of what I should try?

jfedor2, 2024-09-28:
That device just doesn't work currently. It's an Xbox One controller, not a class-compliant USB HID one. We do support some Xbox One controllers like the Xbox Adaptive Controller (and I think the regular first-party ones too), but some of them require custom initialization for some reason.

I have some USB traffic dumps for the new Xbox Lite SE, but I don't have the controller itself so I haven't made any attempts.

CrowKing63, 2024-09-29:
Thanks for letting me know, it's quite a relief to know why it's not working.

closed 2024-09-29 by CrowKing63

jfedor2, 2024-10-24:
So good news and bad news, I got the Xbox version of the 8BitDo Lite SE to work, but only on dual Pico based HID Remappers.

There's a test build of the firmware you can try here:

https://github.com/jfedor2/hid-remapper/actions/runs/11502343575

CrowKing63, 2024-10-25:
Thank you for always being so attentive! 
I'll try it as soon as I get a pico board, thank you very much.

jfedor2, 2024-10-29:
Actually scratch that, in the latest release (r2024-10-29) the Xbox version of the 8BitDo Lite SE should work with both single and dual variants. I noticed it's sometimes finicky when using the wireless dongle, but should be fine with a USB cable.

CrowKing63, 2024-10-30:
Aha! I see, I'll try it as soon as possible then, thanks sir!

CrowKing63, 2024-10-31:
I uploaded the latest released remapper_waveshare_rp2040_pizero firmware and tested it. For some reason, I don't see anything on the monitor tab. I tried both wireless dongle and wired... I even changed the cable just in case, but no luck. When I plug the cable into the pad, the green light above the pad lights up, but then goes out, and when I press the button, it comes back on. I also tried my old gamepad under the same conditions and it was recognized correctly. I guess I'll have to get another board. I'm sorry I can't give you good news after all your hard work. I'll be in touch.

jfedor2, 2024-10-31:
One thing I noticed about the RP2040-PiZero is that it only works if I plug a device into it first and then connect the PiZero to the computer, not the other way round.

CrowKing63, 2024-10-31:
Oh, really? That's really important information. Thanks! I'll try again.

CrowKing63, 2024-10-31:
Umm... again, no.
But I've had the same problem with numeric and cheap keyboards before, so I'm thinking it's my board.

jfedor2, 2024-10-31:
I can't get it to work with the Waveshare RP2040-PiZero either. As mentioned elsewhere it has pull-up resistors on the USB data lines which seem to be messing with stuff.

jfedor2, 2024-10-31:
I can sometimes get it to work with the RP2040-PiZero using the wireless dongle.

jfedor2, 2024-11-01:
I don't know, do you have a soldering iron? If not, getting one will cost more than the Feather probably.

CrowKing63, 2024-11-01:
Hmmm, I have one, but I'll have to see if I can get my family to try it out since I have limited mobility. Pico is definitely a lot cheaper, and feathers take forever to ship.