jfedor2/hid-remapper #180: Only three remap slots possible with Bluetooth?

LardTom, 2024-09-22:
Hey so I have the Seeed Studio Xiao nRF52840 and i want to remap a few keys, i've noticed that only the first three mappings work. why?

jfedor2, 2024-09-22:
There's no such limitation. What device are you connecting and what mappings have you defined?

LardTom, 2024-09-22:
I have the MX Keys and the MX Master 3 connected via Bluetooth. It does not matter what mappings I have, since they just stop working after the third one. I have tried several different mappings, and all of them stop working after mapping three.

jfedor2, 2024-09-23:
I'm not seeing anything like that on my side so either post your config or I can't help you.

LardTom, 2024-09-23:
Here is my config, this is an example since nothing works after the third. Just to clarify I got an Seeed Studio Xiao nRF52840 and flashed it with the corresponding firmware mentioned in the description. everything else seams to work.

[problem-config.json](https://github.com/user-attachments/files/17096326/problem-config.json)

LardTom, 2024-09-23:
it outputs an "a" it is as if there was no remap

jfedor2, 2024-09-23:
And when you go to the Monitor tab and press A, what does it say?

LardTom, 2024-09-23:
It says "A"

[image]

LardTom, 2024-09-25:
Any updates on this?

jfedor2, 2024-09-25:
Not really. My guess is your keyboard was connected to your PC directly when you did the testing and the first three mappings just happened to be for your mouse. I cannot reproduce anything of the sort.

LardTom, 2024-09-25:
i am 100% sure that is not the case, is there a way for me to give you more information?

LardTom, 2024-09-25:
so after experimenting i found out that it isnt neccesarely after the third. it is super weird. when i add more remaps where nothing gets remapped (nothing -> nothing) it seems to work sometimes as if the config does not get send fully. could this be because i use a shitty cable or something?

LardTom, 2024-09-25:
here is the config it works but is not a solution since it does not work every time
[weird.json](https://github.com/user-attachments/files/17135948/weird.json)

jfedor2, 2024-09-25:
What do you see when you click "Save to device" and then "Load from device"? Do mappings disappear?

LardTom, 2024-09-25:
> What do you see when you click "Save to device" and then "Load from device"? Do mappings disappear?

no it is the same

jfedor2, 2024-09-25:
Then it seems they're saved correctly. Though if you do it immediately then they're loaded from memory so to test for any flash storage issues you can save, then unplug and re-plug, then load and see if they're all there.

jfedor2, 2024-09-25:
Do you also double click on "save"? You're not supposed to.

LardTom, 2024-09-25:
> Do you also double click on "save"? You're not supposed to.

sometimes yeah good to know

LardTom, 2024-09-25:
ok so after unplugging and re-plugging it seams to work. so everytime i want to update the rebindings i have to replug it?

jfedor2, 2024-09-25:
No, you shouldn't have to. That was just to test.

LardTom, 2024-09-25:
> No, you shouldn't have to. That was just to test.

well as i said right after replugging the device it works directly

LardTom, 2024-10-05:
> No, you shouldn't have to. That was just to test.

are you looking into that issue? it is kinda annoying

jfedor2, 2024-10-05:
I've added some protection to the configuration website to prevent you from clicking the save button before it's done saving and a confirmation once it's done. Other than that I'm not really experiencing the issues.

LardTom, 2024-10-06:
> I've added some protection to the configuration website to prevent you from clicking the save button before it's done saving and a confirmation once it's done. Other than that I'm not really experiencing the issues.

can it be because of the browser or something?