jfedor2/hid-remapper #181: Waveshare RP2040 Pi Zero not accepting anything via USB-PIO port

domiluci, 2024-09-23:
Bought a [Waveshare RP2040 Pi Zero](https://www.amazon.com/dp/B0CJ6VM6H6?ref=ppx_yo2ov_dt_b_fed_asin_title) the other day to use with an old USB trackpad I have. The firmware upload went smooth, the board shows up fine as "HID Remapper ACHB", but there's no input from the USB trackpad plugged into the USB-PIO port. In fact, there's no power whatsoever going to the trackpad.

I tried a slew of different USB adapters with the trackpad, same result. I tried plugging the Pi Zero into different USB ports, same result. So then I think the keypad must be drawing too much current or something, despite barely drawing anything, so I hook up a plain, B flat [USB numpad](https://www.amazon.com/dp/B0BNPVMQVT?ref=ppx_yo2ov_dt_b_fed_asin_title) instead and, you guessed it, same result. No power to the device on USB-PIO. No input on HID Remapper.

I have stuff to build a RP2040 HID Remapper, but I don't have the physical means to assemble one easily anymore. I was hoping the Pi Zero would've been an easy plug-n-play solution so I could avoid soldering and assembly, plus the header makes GPIO additions and expansions easy. I'm hoping there's still a redeeming fix here to get it working, but I can't find one.

Any help you could give would be greatly appreciated! Thanks!

jfedor2, 2024-09-23:
Did you flash the correct firmware (remapper_waveshare_rp2040_pizero.uf2)?

Can you try with some other devices like your regular keyboard or mouse?

domiluci, 2024-09-24:
@jfedor2 Yessir, firmware is right. I even reflashed it just to be sure. Same issue.

But I don't have anything else I can try right now in the USB department. I'm disabled and can't get down on/up off the ground to get to the cables for my keyboard and gamepad, but both devices are RGB and draw like 10x more current, at least. I doubt they'd power on.

I just ordered a replacement Pi Zero from Amazon. Won't arrive until Wednesday. Hopefully it's just a faulty board. Will update Wednesday/Thursday, and try and find something else to test with in the meantime.

jfedor2, 2024-09-28:
One thing I noticed about the RP2040-PiZero just now is that the order of plugging things in seems to matter, where it doesn't in other cases.

If I plug a mouse or a keyboard into the RP2040-PiZero and then connect it to the computer, it works, but when I first connect the RP2040-PiZero to the computer and then plug a mouse or keyboard into it, it doesn't work. Not sure why.

sonik-br, 2024-09-30:
PiZero works perfectly fine if you connect the usb port to another set of gpio pins. The USB-PIO pins on the board (GPIO06/07) does not work correctly. I think it's related to the [resistors](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/issues/442#issuecomment-2275670753)?

jfedor2, 2024-09-30:
The 22 ohm series resistors are fine, but I'm not sure why they put the other two there (the pull-ups).

Of course the whole point of using this board was to avoid soldering or messing with wires at all so if you're going to wire the USB port to some other pins then you might as well just use a Pico.

It should be possible to cut the traces going to 3.3V, but again, at this point just use the Feather.

I should probably stop recommending the RP2040-PiZero as an option.

sonik-br, 2024-09-30:
> Of course the whole point of using this board was to avoid soldering or messing with wires at all so if you're going to wire the USB port to some other pins then you might as well just use a Pico.

Exactly. I've got one to test as the Feather USB Host is not available here (also not on aliexpress).
Got to the conclusion that PiZero is a no go.

domiluci, 2024-10-03:
> ...at this point just use the Feather.
> 
> I should probably stop recommending the RP2040-PiZero as an option.

Well, I sent the PiZero back and ordered two Adafruit RP2040 Feathers with USB Host built-in. I was stoked when they got here, but now I'm getting genuinely irritated. It's the _same problem as the PiZero_. Whatever you plug into the Feather's USB Host port **_will NOT power on!_** I've flashed the appropriate firmware multiple times on both boards, but the firmware is accessible and functional. I've tried working with the Feathers on two different PCs; both PCs see the Feathers, I can access and change HID Remapper settings using the web tool, but there's nothing that shows up under "Monitor" because the USB device just won't power on. I've even tried my XAC _(with DC power)_ just to see if it's some BS PC USB power issue, but no USB Host power there, either. And I even found some other USB HID devices/mice/keyboards that I've been testing with, too. They work when used normally, but not a single one of those USB HID devices will power on when plugged into the RP2040 Feather's USB Host port.

I'm disabled, but I'm very far from being an idiot, which is an annoying assumption most people make at first glance these days just because my issues have now become visibly apparent. The fact is that I've been working with electronics for over two decades. I earned degrees in comp sci and engineering sci by the time I was 19 years old. And I have legitimate, honest-to-God patents on several of my creations. So, in short, I think I'm at least marginally qualified enough to troubleshoot a ready-to-go, plug-n-play system like HID Remapper, especially if the RP2040 Feather has worked for you in the past. And I _have_ gone through the efforts of troubleshooting, and beyond, but it's still not working and I genuinely cannot figure out why.

@jfedor2: What was the last release of HID Remapper you know, for sure, was tested and working with the RP2040 Feather boards? How about the RP2040 PiZero boards?

jfedor2, 2024-10-03:
The latest release (r2024-06-27) works on both boards, with the caveats mentioned above for the RP2040-PiZero.

I tested the XAC with the Feather just now and it works for me.

sonik-br, 2024-10-03:
@domiluci I don't have any experience with the Feather USB Host.
But I do know that it needs a gpio pin state to send power to the usb host port.
https://github.com/jfedor2/hid-remapper/issues/181#issuecomment-2391390277

Flashed the "remapper_feather.uf2" file?

Please keep in mind that, like I said, I've never used a Feather USB Host before. I'm just trying to help :)

domiluci, 2024-10-03:
> Flashed the "remapper_feather.uf2" file?

Yes I did. On both boards, several times, as mentioned. The firmware works fine. I can read/write to it perfectly with the web tool. But the device plugged into the USB Host port, whatever device I try, fails to power on.

> But I do know that it needs a gpio pin state to send power to the usb host port.

Well, @jfedor2 just mentioned that the latest release works for him and his RP2040 Feather, which leads me to conclude that the relevant GPIO pin was properly set in the firmware before compilation. So we're back to square one. Unless I just got two bad boards and/or all five USB devices I'm trying happen to be the only generic HID devices on the planet that just don't work with HID Remapper (which I find unlikely in both cases).

domiluci, 2024-10-06:
Still playing with this and still can't get it to work no matter what's plugged into what. I'd really like to resolve this considering how financially invested I am in this now, and the fact Adafruit won't accept returns on opened items. So if anyone out there has a fix, I'd love to hear it.

sonik-br, 2024-10-06:
@domiluci can you code?
You could do simple tester to just test if the usb-pio is working. I can help with that.

domiluci, 2024-10-07:
> @domiluci can you code? You could do simple tester to just test if the usb-pio is working. I can help with that.

I can. But usually I just ChatGPT the hard stuff and fill in the blanks. However, I flashed the boards with a TinyUSB HID Remapper example and they still didn't work. I'm pretty sure it's some kind of enumeration issue because USB-PIO is, definitely, getting power.

**_BUT_** this morning I had (some) success! Finally! I purchased a [Darmoshark SK16 macro keypad](https://www.amazon.com/dp/B0CYBRPL87) on Amazon. It's a mini QMK keyboard with BLE/USB/2.4G functionality. A pure HID device. I remapped some of the buttons to give me a WASD/LRUD layout, plugged it into the RP2040 Feather and, whaddya know? The SK16 powered on and HID Remapper recognized it. Which begs the question, out of seven different USB HID devices, _what's special about the SK16, and what's preventing HID Remapper from working with the other six HID devices?_ Might be worth looking into.

There's another issue I've run into, however. It's more like a really bizarre bug. I was remapping some of the keys to Xbox's ABXY and D-Pad buttons and discovered they're mapping incorrectly and/or not at all.

They're mapping out as follows (_Mapped Key_ = _Actual Output Key_):
- X1 = A
- X2 = B
- LSB = X
- LB = Y
- A = LB
- B = RB
- X = LS Click\LSB
- Y = RS Click\RSB
- RB = Menu
- RSB = View
- View = LT
- Menu = RT
- LSX\Y maps okay with expressions. I got it working fine
- RSX\Y I have yet to try, but it's likely okay too
- D-Pad doesn't work at all. I'm using one of the example expressions. All it does is modify AXIS9 on Joypad.io.

I pulled most of my setup from the `Keyboard to XAC` example on the web config tool. I didn't modify any of the expressions and they work, except for the D-Pad. I only changed the key mappings in HID Remapper to account for the strange mappings. And, yes, I have it set to XAC/Flex in the HID Remapper settings. So I'm not sure what's up there.

sonik-br, 2024-10-07:
> The SK16 powered on and HID Remapper recognized it.

Interesting.

Btw I've seen various devices with different behaviors.
This happens on all RP2040 boards I've tested.

* Some devices only works if connected to native usb port.
* Some devices only works if connected to pio emulated port.
* Some devices works on both but only enumerates if connected to pico before the pico being powered up.

The native port when used as usb host works with almost everything but I do have one device that does not work.
And I've tested four devices that simply does not work at all when using PIO. So yeah, PIO is not perfect.

There's a two board version of hid-remapper that uses the native port as host.
You could try this setup as I believe you now have two RP2040 boards, right?

domiluci, 2024-10-12:
> Is your XAC on the latest firmware (at least 5.21.3242.0, released in June 2024)?
> Is HID Remapper plugged into the left USB port on the XAC?
> If you have anything remapped on the XAC itself then of course the inputs will no longer match.

@jfedor2 Yup. And everything's up to date, and all the XAC mappings are unchanged. But I should clarify that this only happens when the RP2040 Feather is plugged into my PC (when updating mappings), not the XAC. And yes, HID Remapper is set to XAC/Flex mode, and I power cycle the board after saving and before testing. But when I go to Joypad.io or GPV to check the mappings, before swapping the Feather from my PC to XAC, they show up incorrectly with the aforementioned mappings.

_However,_ when I then plug the same RP2040 Feather with the same HID Remapper mappings into the XAC, and then check my mappings again with Joypad/GPV, they show up correctly. So I'm not exactly sure what's up there. It works in the end, but it's an odd bug I thought I'd mention. It just makes things a bit confusing LOL 😕

> There's a two board version of hid-remapper that uses the native port as host.
> You could try this setup as I believe you now have two RP2040 boards, right?

@sonik-br Yeah, I actually have several RP2040 boards now. I got a bunch of these [RP2040 Zero](https://www.amazon.com/gp/product/B0DGH2W61Z) boards, among others. I was thinking of making a dual RP2040 HID Remapper contraption out of a couple of them. Maybe a PCB if it works. We'll see 😄!

jfedor2, 2024-10-12:
The button labels are not a part of the USB spec. On some gamepads button 1 is "X", on others it's "A" or "square" or whatever.

If you want to increase the chance of your PC getting the right button map, use the "Switch gamepad" or "Stadia controller" modes, as things like Chrome and Steam have the correct mappings for them hardcoded.

The "XAC/Flex compatible" mode is for plugging HID Remapper into the XAC (and the Hori Flex).

domiluci, 2024-10-15:
Yes, I'm thoroughly aware it's for plugging into the XAC/Flex, and that button labels aren't part of the USB spec. I was just trying to be of assistance and mention something I thought was odd and potentially buggy. But that's fine, and it really doesn't matter at this point. I got it working well enough to hold me over until I finish a more permanent, all-in-one ground-up build.

Thanks for the help.

closed 2024-10-15 by domiluci