jfedor2/hid-remapper #182: Feature request: More buttons and axis on controller output emulation.

A-C-18, 2024-10-16:
Would it be possible to increase the amount of buttons that can be used as an output? The limitation of the controller buttons is to small, for example I have a device with over 40 buttons and 4 axis. If the existing controllers are intended to not change, then maybe consider to add a new controller type with maybe 5 or more axis, and around 50 to 60 buttons. That would also enable the remapper to be easily used with custom build devices. If this is to far from what the remapper should be, then maybe consider to add a device option that combines mouse, keybord and joystick.

jfedor2, 2024-10-16:
Probably best to compile a custom version of the firmware tailored to your needs.

Adding more buttons and joysticks is a fairly simple change in the report descriptor.

A-C-18, 2024-10-17:
I have never before worked with usb devices and descriptors. Could you point me in the right direction for what to look into?

A-C-18, 2024-10-20:
Okay, so i fiddled with the code a bit and the outcome is not quite as expected. I probably missed a few important points... 

0x05, 0x09,                    //   Usage Page (Button)
    0x15, 0x00,                    //   Logical Minimum (0)
    0x25, 0x01,                    //   Logical Maximum (1)
    0x75, 0x01,                    //   Report Size (1)
    0x95, 0x1F,                    //   Report Count (39)
    0x09, 0x01,                    //   Usage (0x01)
    0x09, 0x02,                    //   Usage (0x02)
    ...
    0x81, 0x02,                    //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x75, 0x01,                    //   Report Size (1)
    0x95, 0x01,                    //   Report Count (1)
    0x81, 0x01,                    //   Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
    
That is what i have in the file our_descriptor.cc, the "..." stands for the rest of the numbers. I excluded 0x08 because of its predefined function.
The problem i have is that i can't sett the output on the website to some of my defined numbers and the passthrough works just with a few. I also tried setting the output in the .json file, but that didn't work either. I hope this question isn't out of place here...

jfedor2, 2024-10-20:
You seem to be on the right track.

The buttons you add won't have proper labels on the configuration website, but you should still be able to select them after you connect to your HID Remapper, they should be at the bottom with just hex codes like 0x0009000a etc.

You probably also have to update the *_neutral array for the descriptor you're modifying (add zeros in the right place for the extra buttons).

Other than that the only thing to worry about is that the entire report has to end on a byte boundary (size in bits divisible by 8), but you seem to have that covered (though beware of comments not matching the actual values in the report descriptor - for example 0x1f isn't 39!).

Here's something that works for me: https://github.com/jfedor2/hid-remapper/commit/309ce8c9589c754b8ea156cad6973571350169ca

A-C-18, 2024-10-22:
Thank you for your answer, the problem was really the _neutral array, got it working now.