jfedor2/hid-remapper #183: Example request: set/reset of drag scroll layer dependent on v scroll rotation

itsnoteasy, 2024-10-19:
I find that my hand is developing rsi from holding the drag scroll momentary button, so i decided that i would use the sticky toggle in combination with the vertical scroll function of the slimblade pro. this is because it does not require me to hit any switches.
However i discovered that by rotating the ball it toggles the drag scroll on off on off. this is annoying because it acts like a wheel of fortune.
If i could turn the ball clockwise to set the drag scroll layer on and turn the ball anti clockwise to reset the drag scroll layer to off, then i'd be happy.

jfedor2, 2024-10-20:
Try something like this:

Mapping:

Register 1 -> Layer 1

Expression:
```
1 recall
0x00010038 input_state -1 eq bitwise_or
0x00010038 input_state 1 eq not mul
1 store
```

Twisting the ball one way should activate the layer and twisting it the other way should deactivate the layer. Swap the "1" and "-1" to reverse the direction.

Here's a JSON file with the config: [hid-remapper-twist-layer.json](https://github.com/user-attachments/files/17451472/hid-remapper-twist-layer.json)

itsnoteasy, 2024-10-20:
it's perfect thanks. i had i minor issue with scrolls being registered when drag scroll was disabled but i bound v scroll to nothing and it's fixed.

closed 2024-10-20 by itsnoteasy